opentx/opentx #6194: Model ID in use message in D8 mode

kreidler, 2018-09-29:
OTX 2.2.2 x9d+

Message "model id already in use" (sorry for bad translation using german version) appears after successful bind of a D8 receiver.

- Copy a model with D16 mode
- Switch mode to D8 and bind
- After bind messagebox with "model id already in use" appears but there is no possibility to change or remove the receiver number.
- Also reproducible after powercycle of TX and rebind.

ParkerEde, 2018-09-29:
yes. it's the same OTX 2.2.2 Horus X10S

Squirlier, 2018-10-04:
if i remember correct a workaround for this is:
select d16 mode. then a field called "id" od similar comes up which is not shown in d8.
simply change the id to next higer number and switch back to d8.
it would be easy if the id field is shown in d8 mode also.
i am not sure if this works. maybe you have to leave model after changing the id and before you change back to d8 mode. (for saving)

3djc, 2019-05-01:
Done, will be in 2.3

PR #6372 (2019-05-18): Fixes ModelID unduly checked in D8
Thix fixes #6194
closed 2019-05-01 by 3djc