opentx/opentx #6195: Model transfer over wireless

richardgillingham-ensygnia, 2018-09-30:
For transmitters that have wireless linking - horus QX7S it would be great to be able to send a model definition from one tx to another over the wireless link.  It would be fantastic at the field

Luznatural, 2018-11-19:
I fully agree on the need of having a way to transmit data from one radio to another one, without the need of a computer. Fully support...

lshems, 2018-11-19:
Not even Wireless. Would be happy with sdcard backup models being more promoted as way of exchanging models.

It already works flawlessly, but with lack of version and radio type control, which makes it dangerous.

Could be easily improved by including some header in the file.