opentx/opentx #6201: Random LUA errors - string expected, got lightfunction

azaz44, 2018-10-03:
**Describe the bug**
I get such errors randomly, when using a one-time script. They are probably related to `string.format` function, but not 100% sure. This is where I see them most commonly.

Unfortunately it seems that variables or expressions used in these places are of correct types. And what's worse, the problem appears and disappears randomly. Adding a single line of comment somewhere in the code (`-- something`) can make the problem disappear or appear again. Also it works a little bit like quantum physics - once you try to find a problem and print some debug output, the problem does not happen anymore.

More details:
- problem happens on simulator in Companion, simulating Horus X10 radio. I didn't test on the real radio, considering random nature of the bug.
- LUA compilation is enabled

**To Reproduce**
I think I found a small example which seems to reproduce the problem on my machine every time. I hope it's not machine dependent. Running this one-time script gives me the error.
```
datetime = {}

function datetime.formatDate(dt)
  --print("Y = " .. dt.year .. " | M = " .. dt.mon .. " | D = " .. dt.day)
  return string.format("%04i-%02i-%02i", dt.year, dt.mon, dt.day)
end

function datetime.formatTime(dt)
  return string.format("%02i:%02i:%02i", dt.hour, dt.min, dt.sec)
end

function datetime.format(dt)
  local d = datetime.formatDate(dt)
  local t = datetime.formatTime(dt)
  return d .. " " .. t
end

local function init()
end

local function run(event)
  lcd.clear()
  for i=1, 100 do
    local dt = getDateTime()
    local t = datetime.format(dt)
    lcd.drawText(30, 30, t)
  end
  
  return 0
end

return { init=init, run=run }
```

The error is:
```
Script error: //test.lua:15: attempt to concatenate local 'd' (a lightfunction value)
```

Line 1 is the one with `return d .. " " .. t`.

However, uncommenting line 4 (`-- print...`) causes the script to execute just fine.

**General information**
 - OpenTX version: opentx-x10-2.2.2
 - Hardware: sim
 - OpenTX Companion: 2.2.2, Jul 28 2018 running on Windows 10 x64.

azaz44, 2018-10-04:
Updated - just had a similar problem today. It seems `string.gsub` is also affected. For no reason it returns a `lightfunction`. Just adding some `print` in between to debug the problem makes the problem disappear.

azaz44, 2018-10-14:
I'd like to bring more attention to this problem. I'm trying to make some nice LUA development for OpenTX, which you can see on screenshots, and spent quite some time on it, but this problem makes it impossible to continue.

As for the  bug, I made more researche, as it is happening to me right now as a result of calling `string.gsub`. If this function returns a `lightfunction` instead of a string, the function which is returned is exactly `string.gsub` function. I suspect some stack corruption, maybe in `luaL_pushresult`, but I'm not familiar enough with OpenTX source.

Also, it probably happens on real radio too, not only simulator. Not 100% sure as I don't have the console output there, but things are crashing.

[image]
[image]

3djc, 2018-10-15:
Not a solution, but maybe a way to avoid the issue for now, it works if you replace `return d .. " " .. t` by `return  string.format("%s %s", d, t)`

azaz44, 2018-10-15:
> Not a solution, but maybe a way to avoid the issue for now, it works if you replace return d .. " " .. t by return string.format("%s %s", d, t)

Well, it seems any change in the code can make it work, even meaningless change like adding a comment. But then it fails after some other changes and only sometimes.

I'm trying things like this now, but I have yet to verify if it works:
```
-- workaround for the lightfunction problem...
function string2.format(...)
  
  local ret = nil
  local numWorkarounds = 0
  
  while not ret do
    ret = string.format(...)
    
    if type(ret) == "lightfunction" then
      numWorkarounds = numWorkarounds + 1
      ret = nil
    end
    
  end
  
  if numWorkarounds > 0 then
    print("*** *** *** Applied string.format lightfunction workaround " .. numWorkarounds .. " times..")
  end
  
  return ret
end
```

But for now it's just my hope that it will end up working and not in an infinite loop there. So far it seems it only happens with `string.format` and `string.gsub`.

3djc, 2018-10-16:
@projectkk2glider  @mpaperno  Any of you inspired ?

closed 2019-11-15 by stale[bot]

assada, 2023-03-29:
Same on EdgeTX 2.8