opentx/opentx #6206: X7 won't boot with 2.2.2

skorokithakis, 2018-10-05:
I flashed OpenTX 2.2.2 on my Q X7, it loaded fine, worked fine. I turned off the radio, tried to turn on again, the power button just flashes green and nothing happens. If I turn the radio on to the bootloader and then press "exit", the firmware loads and works fine. It just doesn't work when booting normally.

Downgrading back to 2.2.0 works fine.

**General information**
 - OpenTX version: 2.2.2
 - Hardware: Taranis Q X7
 - OpenTX Companion: 2.2.2 on Ubuntu 18.04

3djc, 2018-10-05:
Are you maintaining the power button pushed ?

skorokithakis, 2018-10-05:
Sorry, I was not :( Is this new in 2.2.2, or did I just forget how the radio turns on?

closed 2018-10-05 by skorokithakis

3djc, 2018-10-05:
It is new, a feature that was added to prevent the radio turning on by itself during transport in a backpack for example

skorokithakis, 2018-10-05:
Ah, so I'm not crazy, thank you. It would be nice if this were more advertised, as it's a pretty big change for a minor version...