opentx/opentx #6207: No Trainer on X-lite

g19i85g, 2018-10-06:
I will use the SBus Trainer mode on the Taranis X-Lite but i can not find the menu on the model settings.
Tested with OpenTX 2.2.2.

kilrah, 2018-10-06:
There is currently no trainer functionality on X-lite.

noXi89, 2018-10-12:
There is a trainer function with the bluetooth module installed.
Close?

g19i85g, 2018-10-17:
Can we make a feature request out of it?

Protonus, 2019-05-08:
Anyone find a solution for this?  Looking to slave my X-Lite to DJI Occusync Air.  Just needs a trainer port.

closed 2020-02-02 by stale[bot]

matthewthorp, 2020-04-02:
Trainer option in the model menu missing from opentx. I've recently bought a Bluetooth module to add to it as in this video.
https://www.youtube.com/watch?v=fyT6XohZIsY&t=478s

Gone through version history and seems to be showing last in v2.2.2

General information
 - OpenTX version 2.3.7
 - Hardware Frsky x-lite
 - OpenTX Companion 2.3.7 on windows

kvng1, 2021-12-11:
> So it turned out to be a quite simple mod, a matter of removing two capacitors and jumping two wires, as well as some PCBXLITE(S) amendments. Maybe I'll add speaker/trainer switch circuitry later on as well.

Hello, was you able to get the trainer port working on x-lite? Can you please share some direction?

DieHertz, 2021-12-12:
It required both hardware and software modification, I removed two capacitors which are used to drive left and right headphone channels, and added jumper wires from specific MCU pins to jack connector. This removes the possibility to use headphone plug for headphones unless you also add some switch circuitry, but I don't use headphones so I didn't bother.

I could share more details a bit later if that sounds like a mod you're willing to do l.

kvng1, 2021-12-12:
Yeah, that sounds doable, I also don't need headphone. Are you using the x-lite s firmware or have to build a custom?

DieHertz, 2021-12-12:
A custom one, using Xlite S firmware would be cool but it requires more more, e.g. backlight control needs to be rerouted because FrSky swapped some pins around.
I'll share all changes necessary

DieHertz, 2021-12-12:
[image]
Here's basically what I did, PD12&PD13 are both unused in original firmware, I shorted them just to make soldering easier, you only need one or potentially none, as jack plug event auto-detect does not work reliably with original wiring anyway.
Will share code tomorrow from my work laptop.

kvng1, 2021-12-12:
Thank you, so pc13 for trainer input and pd12&13 for jack plus detection. Is there any unused pin on the female jack that I can jump the pc13? In this case, I need an adapter but headphone is still functional.

DieHertz, 2021-12-12:
They hardwired the only potential jack pin to ground, so it's not possible without unsoldering connector and doing some heavy-duty modding I think

DieHertz, 2021-12-14:
https://github.com/DieHertz/opentx/tree/hertz-xlite-trainer-jack-support-2.3

kvng1, 2021-12-14:
Thank you for sharing it. I'm comfortable with hardware mod but building custom firmware hold me back a bit, look like you have to build a new one whenever a new version is introduced. I'm reading some build guide now but it'd take a while.

DieHertz, 2021-12-16:
That's a necessary evil I'm afraid, I can build it for you if you want.
I personally update my radio firmware very rarely.

Dragony, 2022-01-08:
@DieHertz I'm looking to support SBUS Trainer support on my X-Lite S. From what I can see in the code, I can use the same hardware mod you posted above. Can you confirm?

DieHertz, 2022-01-09:
No idea, I don't have S, only original X-Lite

Maze1982, 2023-09-09:
@DieHertz I have tried but i just not have what it takes to build it for myself.
Could you help med also? 

/Maze

DieHertz, 2023-09-10:
Hi, could you remind me what exactly we're taking about here? You did the mod and need a patched firmware?

DieHertz, 2023-09-10:
No problem, feel free to ask me if you need further help.

Sorry @jereitor I could not answer you, it's been a busy year

ligteltelecom, 2024-06-03:
Hi i have frsky xlite (old version) and need add trainer port. Is this your mode is compatible with it? PC13 and PD12& 13

DieHertz, 2024-06-03:
My X-lite was the latest pre-production version, I guess it should work, but I can't check anything as I no longer have access to any hardware

ligteltelecom, 2024-06-03:
> My X-lite was the latest pre-production version, I guess it should work, but I can't check anything as I no longer have access to any hardware

Hello, which files do I need to download from your github? The hardware part, I have no doubts, is very clear to me. but I didn't find any information about what I should change in the software.

DieHertz, 2024-06-03:
You have to manually build the firmware from the branch mentioned in this message:
https://github.com/opentx/opentx/issues/6207#issuecomment-993362949

ligteltelecom, 2024-06-03:
> You have to manually build the firmware from the branch mentioned in this message: [#6207 (comment)](https://github.com/opentx/opentx/issues/6207#issuecomment-993362949)

Hello, do you have any instructions on how to build the firmware with your files?

DieHertz, 2024-06-04:
> > You have to manually build the firmware from the branch mentioned in this message: [#6207 (comment)](https://github.com/opentx/opentx/issues/6207#issuecomment-993362949)
> 
> Hello, do you have any instructions on how to build the firmware with your files?

It's no different from building OpenTX in general, which is not that straightforward.
I would follow this commend and recommend the Docker way as it has fixed and isolated dependencies:
https://github.com/opentx/opentx/issues/8528#issuecomment-855251520