opentx/opentx #6214: XLite allow dpad to be used as momentary switches

rodgerramjet, 2018-10-09:
Wondering if this is possible- I’m a miniquad / fpv pilot and we have zero use for manual trim adjustment. 

Is it possible to make this a configurable option to give me 4 more buttons, if I don’t need the trims.

Each direction should be its own button/ channel.

I know you can do this with some special settings but that’s pretty clunky to do and doesn’t give all features we’d need

kilrah, 2018-10-10:
They already are, there are several tutorials around for this.

closed 2018-10-10 by kilrah

expipiplus1, 2019-02-19:
@kilrah, would you be able to post a link to one of these tutorials, please?

expipiplus1, 2019-02-19:
One way I've found to do this is to:

- Disable the trim in the flight mode(s) you will be using
- Create two special functions for `<trim symbol><trim number>u` and `<trim symbol><trim number>d` which override a channel to the desired values for trim up and trim down
- Be careful not to override any channel you are using elsewhere