opentx/opentx #6215: X9D: selection Source not possible for calculated Telemetry Values (OTX 2.2.2)

r41065, 2018-10-10:
Building a new model on my Taranis X9D ( no Plus, OpenTX 2.2.2 ) 
i want to set new telemetry values for each cell of my FLVSS sensor:
- Discover new sensor: Cels is found.
- Then add new calculated sensor for each cell:
Selection menu for source shows only ---, not the discovered sensors..

As a workaround i set the source to single cell values in Companion and transfer it back to the transmitter but editing the source at X9D is not possible. ( even not for models which are converted from OTX 2.2.1 )

r41065, 2018-10-10:
Yes, done in same way as i've done it with Version 2.2.1 directly on X9D 
( was working for about 42 models with FLVSS and MLVSS sensors, so i wonder why not...)

add new sensor
  give name ( Z1)
    select calculated
      try to select source: selection menu shows only ----
select Cell Value ( first, second, lowest ....) and check marks for logging etc. is possible again...

Selection this new calculated Value on telemetry screens is also possible ( but value is only blinking 0.00)

3djc, 2018-10-10:
Could you be more specific ? I'm trying to follow your sequence and I have the sources

[image]

r41065, 2018-10-10:
First picture: Working sensors edited from Companion:
[image]

Trying to set 3rd Cell
[image]

Select Source  "Cels" ist not possible...  looks like it is only not possible to formula CELS, i didn't try the others before....

kilrah, 2018-10-10:
Try to delete and rediscover your Cels sensor.

r41065, 2018-10-10:
OK I Try when im home again ( maybe next Weekend...)
If it was to the converted Models i agree with you but this was a new designed Model and it was the first time i had discovered the FLVSS...

closed 2018-10-10 by r41065