opentx/opentx #6218: Graph in INPUT screen does not represent the current edit

notlebowski, 2018-10-11:
2.2.3N680

When an input is edited we get a graph that represents the curve.
If that input is one of several inputs and the edited one is not enabled (through switch or flight mode), I'm getting the graph of the active one.
I wouldn't expect that and maybe that's the bug already.

Maybe it is intended that one should see the graph of the active one and not the one that's being edited. I hope that's not the case.
But if I change, for instance, the source the graph then suddenly changes into something that's not meaningful.

I have this example and SA is down:
```
Inputs
I1:Ail
    Ail Weight(+20%) Switch(SA↑)
    Ail Weight(+100%) Expo(50%)
```

[image]

[image]

kilrah, 2018-10-11:
> Maybe it is intended that one should see the graph of the active input and not the one that's being edited. I hope that's not the case.

It is the case, it's how several other radios work so is something people are used to.

notlebowski, 2018-10-11:
OK
But what about what happens if you change the source?

kilrah, 2018-10-11:
Seems there's an issue with that indeed. And it must not be something people look at often because it's been there forever aka >4 years.

notlebowski, 2018-10-11:
Normally I wouldn't go there, but I started fiddling with it to make sure it was indeed intended behaviour that the active graph was shown.

The intended behaviour is alright by me.
As long as you know, it's alright. I now know....  ;)

notlebowski, 2018-11-04:
You may have been misled by the title of the ticket, but there was indeed something wrong with the graph of the input screen and I wouldn't have made this ticket if it was only its behaviour that it didn't represent the current input at hand.
I added that as I wanted to make sure it was intended behaviour. 
What happens after changing the source was not intended behavior.

closed 2019-02-25 by notlebowski