opentx/opentx #6223: X-Lite and R9M flex

skaman82, 2018-10-17:
I am not sure if it is a bug, but I've just installed the newest nightly available for my x-lite on the Companion and flashed my R9 mini module with the latest flex firmware from frsky. I can change the frequency but not the power output or sport setting. The power output is always "(100mw)" 
How I can toggle between 25 and 100mw?
[image]

kilrah, 2018-10-17:
Power is not adjustable on the Lite module.
https://www.frsky-rc.com/frsky-r9-series-new-products/

closed 2018-10-17 by kilrah

skaman82, 2018-10-20:
Ok, thanks @kilrah. But why it is (100mx) and not 100mw. Does it mean something?

3djc, 2018-10-20:
( ) is actually to show it is information only, and not a value that can be changed

skaman82, 2018-10-20:
Then s.port ON value should be (ON) too, because it is also not changeble, right?

3djc, 2018-10-20:
Unlike RF power that cannot be changed, telemetry value can be changed depending on internal module status