opentx/opentx #6225: FrSky drivers not compatible with USB3.0

WynandRedelinghuys, 2018-10-22:
There seem to be a problem with the Companion2.2.2 device driver files. It initiates and runs without a problem with USB2. As soon as the Intel(R) USB 3.0 eXtensible Host Controller is loaded the drivers stops working. By Using the MicroSoft mass storage rather than the Taranis mass storage driver the drive is open up correctly. 

 FrSky_Taranis_Mass_Storage and Taranis Create Device Drivers initiates the USB3.0 port but does not show the Taranis QX7 radio or the SdCard as drives.

kilrah, 2018-10-22:
There are no drivers that you're supposed to use for mass storage but the original microsoft ones.

closed 2018-10-22 by kilrah

WynandRedelinghuys, 2018-10-22:
Open-Tx 2.2.2 comes with its own device drivers for the Taranis x7 radio and is automatically installed when installing Companion2.2.

Is is easy enough to replace the - FrSky mass-storage  driver for the SDCard - with that of MicroSoft. 

The problem arises with the - FrSky driver for the radio's EEPROM. It doesn't have a problem with USB(2.0) but does not work with USB(3.0). 

It will be much appreciated if someone can help. I had a look at some of the other posts and it seems to be a regular topic. "No Radio Found." The USB driver (3.0) initiates as a device, but the radio's EEPROM is not available as a drive. No data can thus be copied to the directories in the radio's EEPROM and Companion2.2 does not see the radio.

There is no problem up and down loading the firmware using DFU-UTIL.

schwabe, 2018-10-22:
It does not come with driver as @kilrah said. OpenTX on the device emulates a regular mass storage device.

WynandRedelinghuys, 2018-10-22:
Now you have completely lost me. None of the drivers below was on my computer before I installed OpenTX

 [image] 

Do you perhaps know who is the driver provider for libwdi with a generator code of USB\VID_0483&PID_5720 (libwdi autogenerated) is? 

[image] [image]

Your support will be highly appreciated. Companion2.2 will not allow me to simulate or upload the radio on the computer through USB(3.0) - "No Radio Found."

kilrah, 2018-10-22:
As mentioned the libwdi drivers were installed by you using zadig incorrectly.

WynandRedelinghuys, 2018-10-22:
Andre I am now completely lost. I have uninstalled and deleted all STM32 Bootloader and Frsky Taranis Mass storage devices and files from my computer. 

When the radio is switched off and I plug it into the USB port the STM32 Bootloader is loaded as a driver.

If the radio is switched on and I plug it into the USB port the options - USB Joystick (HID) and - USB Storage (SD) appear on the radio. By selecting the USB Joystick the Taranis radio appears as a FrSky joystick under the devices. By selecting USB Storage the FrSky Taranis mass storage driver is installed.

The installation of these drivers must be initiated from the Taranis radio. I did a search afterwards on the computer but could not find any of the drivers. I also was disconnected from the internet. There is no change in Companion. Still don't see the radio as a virtual drive and with  the -Taranis mass storage driver- installed the virtual SD card also disappeared. The radio needs a driver to communicate with the USB. Don't you think there is a bug in the USB Descriptor file file. It runs perfectly under USB(2.0) but crashes when using USB3.0

kilrah, 2018-10-22:
Did you check the "remove driver" checkbox when uninstalling as requested? If you didn't then the wrong driver isn't removed, stays on your system and will be reused next time you plug.

WynandRedelinghuys, 2018-10-23:
[image]

Uninstalled and deleted all the Taranis device drivers. No access to the radio. Re-installed Companion. No access to the radio - cannot up/download firmware. Replaced libwdi (default installation) with USB mass storage - SDCard virtual drive is up and running.

Thank you for the help.