opentx/opentx #6226: Compiling OpenTX on Win10 PC

LapinFou, 2018-10-24:
Hi folks,

First of all, big big thanks to @mpaperno !
I followed the instructions from this wiki [OpenTX 2.2 Build With QtCreator (Win Linux OSX)](https://github.com/opentx/opentx/wiki/OpenTX-2.2-Build-With-QtCreator-%28Win-Linux-OSX%29)
Indeed the installation is much simpler.

Using SourceTree, I checked out the tag 2.2.2

I got some small issue to install **pillow** in Pyton 2.7.15.
If will update the documentation for Windows user as soon as I'm able to build the project on my Win01 laptop.
I will also add some comments how a where to install SDL lib on Windows machine. 

So far, I got this in QtCreator:
```
Running Windows Runtime device detection.
No winrtrunner.exe found.
Running "C:\Program Files (x86)\CMake\bin\cmake.exe C:/Users/Sebastien/Downloads/github/OpenTX "-GCodeBlocks - MinGW Makefiles" "-DCMAKE_CXX_COMPILER:STRING=C:/Qt/Qt5.7.1/Tools/mingw530_32/bin/g++.exe" "-DCMAKE_C_COMPILER:STRING=C:/Qt/Qt5.7.1/Tools/mingw530_32/bin/gcc.exe" "-DCMAKE_PREFIX_PATH:STRING=C:/Qt/Qt5.7.1/5.7/mingw53_32" "-DQT_QMAKE_EXECUTABLE:STRING=C:/Qt/Qt5.7.1/5.7/mingw53_32/bin/qmake.exe"" in C:\Users\Sebastien\Downloads\github\build-OpenTX_2.2.2.
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0023 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- OpenTX 2.2.2 @a5de713b
-- Python found, version: 2.7.15
-- Qt Version: 5.7.1
-- SDL Lib: C:/Programs/lib/libSDL.dll.a; Headers: C:/Programs/include/SDL; Version: 1.2.15
TARGET simu/libsimulator: cpp compiler C:/Qt/Qt5.7.1/Tools/mingw530_32/bin/g++.exe v5.3.0
CMake Warning at radio/src/tests/CMakeLists.txt:45 (message):
  WARNING: gtests target will not be available (check that GTEST_INCDIR,
  GTEST_SRCDIR, and Qt5Widgets are configured).

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/Compiler/GNU.cmake:47 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "0" will no longer be dereferenced when the policy is
  set to NEW.  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/Compiler/GNU-ASM.cmake:6 (__compiler_gnu)
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:22 (include)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/Compiler/GNU.cmake:47 (if):
  if given arguments:

    "NOT" "APPLE" "OR" "NOT" "CMAKE_ASM_COMPILER_VERSION" "VERSION_LESS" "4"

  An argument named "0" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/Compiler/GNU-ASM.cmake:6 (__compiler_gnu)
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:22 (include)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeInitializeConfigs.cmake:30 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "0" will no longer be dereferenced when the policy is
  set to NEW.  Since the policy is not set the OLD behavior will be used.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:70 (cmake_initialize_per_config_variable)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeInitializeConfigs.cmake:30 (if):
  if given arguments:

    "NOT" "Debug" "STREQUAL" ""

  An argument named "0" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:70 (cmake_initialize_per_config_variable)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeInitializeConfigs.cmake:30 (if):
  if given arguments:

    "NOT" "Release" "STREQUAL" ""

  An argument named "0" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:70 (cmake_initialize_per_config_variable)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeInitializeConfigs.cmake:30 (if):
  if given arguments:

    "NOT" "MinSizeRel" "STREQUAL" ""

  An argument named "0" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:70 (cmake_initialize_per_config_variable)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeInitializeConfigs.cmake:30 (if):
  if given arguments:

    "NOT" "RelWithDebInfo" "STREQUAL" ""

  An argument named "0" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  C:/Program Files (x86)/CMake/share/cmake-3.12/Modules/CMakeASMInformation.cmake:70 (cmake_initialize_per_config_variable)
  radio/src/CMakeLists.txt:461 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

TARGET bootloader: cpp compiler arm-none-eabi-g++ WARNING: COMPILER NOT FOUND!
TARGET firmware: cpp compiler arm-none-eabi-g++ WARNING: COMPILER NOT FOUND!
TARGET companion: cpp compiler C:/Qt/Qt5.7.1/Tools/mingw530_32/bin/g++.exe v5.3.0
CMake Deprecation Warning at companion/src/CMakeLists.txt:332 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Partial install to C:/Users/Sebastien/Downloads/github/build-OpenTX_2.2.2
CMake Warning at companion/src/CMakeLists.txt:406 (message):
  Installer: SDL.dll not found!

-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/Sebastien/Downloads/github/build-OpenTX_2.2.2
```
So, it seems to be OK.
Then, I clicked on the big **Projects** button.
I selected the target **install** as *Build Steps*.
I also selected the target **clean** as *Clean Steps*.
Finally, I have done a `Ctrl+B`
Unfortunately, the magic did not append.
I got this error message.
Any help would be very welcome. 😃 

[image]

3djc, 2018-10-24:
I guess that wiki is mostly to build companion, and you tried to build a firmware, so it failed because the arm compiler was not found

LapinFou, 2018-10-24:
Thanks for helping. 😄 

According the Wiki, the Arm compiler is not required for the Companion & Simulator?
Or, am I wrong?

Here is what I choose in the build options
[image]

3djc, 2018-10-24:
As I said, your log suggest you are trying to build a firmware because your target seems to be 'all', you probably want to narrow that  down to companion22 for example

LapinFou, 2018-10-24:
OK, the chosen target is **install**, it looks very similar to **all**.
I will try the target **companion** as you suggest.

LapinFou, 2018-10-24:
Looks like is working!!!
👍 
What would be the correct target to build the **opentx-XX-simulator.dll**?
The goal is to build the French simulator DLLs.

LapinFou, 2018-10-24:
I tried the target **libsimulator**, but this generated some **libopentx-XXX.simulator.dll** which are bigger and not recognized by Companion.

3djc, 2018-10-24:
I’m closing since this is not an issue. Be careful if you intend to distribute those that options need to be set similar to the supplied libs.

closed 2018-10-24 by 3djc

LapinFou, 2018-10-24:
> that options need to be set similar to the supplied libs
Why ?
Is it for this reason than the generated **libopentx-XXX.simulator.dll** are not recognized by Companion?

3djc, 2018-10-24:
Yes, for consistency. Release is built with those options: 
https://github.com/opentx/opentx/blob/2.2/tools/build-companion-release.sh#L28

LapinFou, 2018-10-24:
First of all, thanks to you guys for helping me. 😃 

@kilrah I forgot to say that I did rename **libopentx-XXX.simulator.dll** to **opentx-XXX.simulator.dll**, but still no luck. 😢 
@3djc Thanks. I will try tomorrow.

mpaperno, 2018-10-24:
Hi Sebastien  :)   When you say "not recognized" does that mean CPN/SIM doesn't find the simulator for that radio during startup?  The DLL has to be in the same folder as the executable you're running (companion.exe). You should not need to rename anything.  How are you starting companion.exe which you built?

The simulator DLLs built with MinGW are not compatible with Companion/Simulator built with MSVC compiler like the "official" OTX version (they need different versions of Qt5Core.dll to run).

To see all the debug messages while CPN is running, you can _Run_ it from within QtCreator -- in the project config make sure the "Run" action specifies companion.exe as the target (it should be an option in the selection list).  Then you can select Run in QtCreator (instead of Build) and all messages will be sent to the QtCreator output console (one of the tabs at the bottom of the window).  This will show you if it is even trying to load the simulator DLL(s), and any errors/etc if it does try but fails.

`all` and `install` targets _shouldn't_ be building the firmware if a compiler wasn't found... could be a "bug" or an oversight.

LapinFou, 2018-10-25:
Hi Max,

The idea is to build the radio DLL files in French, then to replace the official DLL (which are in English).
From what I understand, it seems that with this procedure I can only build DLL for the MinGW version. Which is different of the official OpenTX. 😭 

Concerning the `all` and `install` targets, I got a compilation error if the ARM toolchain is not preset.

BTW, I will soon push some tiny modifications in your tutorial. Few steps are missing for Win users.

LapinFou, 2018-10-25:
Small updated pushed: [Showing 11 additions and 6 deletions](https://github.com/opentx/opentx/wiki/OpenTX-2.2-Build-With-QtCreator-(Win-Linux-OSX)/_compare/16acdff1d6c9ffca95432487eb95412292331f0a...1efd20febde4a236ca9a6b0d86586516c194a397)

kilrah, 2018-10-25:
> The idea is to build the radio DLL files in French, then to replace the official DLL (which are in English).

Then you have to use the "full" VC++ guide.

LapinFou, 2018-10-25:
In order to keep my Win10 "clean", I will probably create a VM with the official OpenTX dev tools.
Thanks to all of you for your support.
👍

LapinFou, 2018-10-25:
Damn... I just setup a VM and I'm half way installing VS Community 2015...