opentx/opentx #6227: Horus x12s,  - throttle channel causes glitches on yaw channel.

quadsrock2018, 2018-10-25:
Hi,

Using horus x12s with opentx 2.2.2, flying quads copters.

Issue:  Throttle channel "3" interferes with yaw channel "4". "mode 2 set up".
when moving throttle up and down, now and then the yaw channel will get some activity. "this causes the quadcopter to spin".

amy one with a horus x12s seen this issue.,  i will check the hardware gimbal to see it the sensor is lose on the left ginbal

i can give more info or a video of the channel monitor, too dark to do now.

thanks

darryl

3djc, 2018-10-25:
Well this may be an actual issue, it is not one of OpenTx, so closing this. RCGroups or OpenRc forums are proably a more suited place

closed 2018-10-25 by 3djc

quadsrock2018, 2018-10-25:
this started as soon as i updated to opentx.