opentx/opentx #6238: no EVT_PAGEUP_FIRST on X10 on LUA

loetefix, 2018-11-09:
EVT_PAGEUP_FIRST function not working for x10.

I modifyed the api_general.cpp @ line 1402

#if defined(PCBHORUS)
	#if defined(PCBX10)
  { "EVT_PAGEUP_FIRST",  EVT_KEY_LONG(KEY_PGDN) },
  { "EVT_PAGEDN_FIRST",  EVT_KEY_BREAK(KEY_PGDN) },
	#else
  { "EVT_PAGEUP_FIRST",  EVT_KEY_FIRST(KEY_PGUP) },
  { "EVT_PAGEDN_FIRST", EVT_KEY_FIRST(KEY_PGDN) },
	#endif
  { "EVT_TELEM_FIRST",  EVT_KEY_FIRST(KEY_TELEM) },
  { "EVT_MODEL_FIRST",  EVT_KEY_FIRST(KEY_MODEL) },
  { "EVT_SYS_FIRST",  EVT_KEY_FIRST(KEY_RADIO) },
  { "EVT_RTN_FIRST",  EVT_KEY_FIRST(KEY_EXIT) },
#elif defined(PCBXLITE)

now it works well.

Is it possible to take it in?

kilrah, 2018-11-09:
> { "EVT_PAGEDN_FIRST", EVT_KEY_BREAK(KEY_PGDN) },
That seems wrong, would have to be same as X12

loetefix, 2018-11-09:
I used Break and not First because it dont work with FIRST.i set a killEvents(event) in the lua script after a if event==EVT_PAGEUP_FIRST and then it works.with a BREAK it jumps back and forward.

loetefix, 2018-11-09:
sorry with  "EVT_PAGEDN_FIRST", EVT_KEY_FIRST(KEY_PGDN) 
it jumps back and forward

kilrah, 2018-11-09:
That is normal and is why the EVT_PAGEUP_FIRST event doesn't exist, since the same key is used for both they will always conflict with each other. 

By doing { "EVT_PAGEDN_FIRST", EVT_KEY_BREAK(KEY_PGDN) } you will trigger an EVT_PAGEDN_FIRST when you release the key whether it was a short or long press so not only the description is now incorrect (it triggers on release and no more on press) but it also triggers when you release "page up" and not just "page down".

loetefix, 2018-11-09:
Ok, but i see no other way to get an easy compatibility for x12 and x10/s lua-scripts. But if you have an idea it would be nice:)
Thanks

loetefix, 2018-11-09:
Now i have a solution found.How about that?
#if defined(PCBHORUS)
	#if defined(PCBX10)
  { "EVT_PAGEUP_FIRST",  EVT_KEY_LONG(KEY_PGDN) },
    #else
  { "EVT_PAGEUP_FIRST",  EVT_KEY_FIRST(KEY_PGUP) },
    #endif
  { "EVT_PAGEDN_BREAK", EVT_KEY_BREAK(KEY_PGDN) },
  { "EVT_PAGEDN_FIRST", EVT_KEY_FIRST(KEY_PGDN) },
  { "EVT_TELEM_FIRST",  EVT_KEY_FIRST(KEY_TELEM) },
  { "EVT_MODEL_FIRST",  EVT_KEY_FIRST(KEY_MODEL) },
  { "EVT_SYS_FIRST",  EVT_KEY_FIRST(KEY_RADIO) },
  { "EVT_RTN_FIRST",  EVT_KEY_FIRST(KEY_EXIT) },
#elif defined(PCBXLITE)

the description is right and it is possible on x10 to recognize if a Page up or Page down is pressed:

kilrah, 2018-11-10:
That could be OK.

> Ok, but i see no other way to get an easy compatibility for x12 and x10/s lua-scripts.

Your script can read what radio it's running on and adapt itself to that.

closed 2020-04-06 by stale[bot]