opentx/opentx #6241: LUA: need getFieldInfo() accept source ID

azaz44, 2018-11-10:
This could be an extension of existing `getFieldInfo()` or a new function. Current this function requires a source name.

This would allow us to enumerate and get information about telemetry sources.

kilrah, 2018-11-10:
IDs won't work since they are not fixed. But telem1-telem32.

Ryanf55, 2019-02-15:
I'm also confused about this. Scanning for telemetry ID's seems to take a lot of CPU time and doesn't always work to do it during in Lua initialization if the radio is powered on before the model. If Lua had access to what sensors were "Discovered" as some sort of lookup table, that would be much easier to work with.

3djc, 2019-02-15:
Would you mind sharing your code so I can take a look at what is happening ? Sure we could have a function that would give the sensors as a table, but essentially, that would be parsing telem1 to telem32 and building that table, which is probably what  you are doing already.

azaz44, 2019-02-15:
I hope I still remember what the issue was about. I'm not sure if you wanted my code or Ryan's, but here's some excerpt from me.

This code is about allowing a user to select a source. The source is drawn with `lcd.drawSwitch()` and user can scroll to prev / next source. `_isSourceValid()` is there to check which sources he can select, so we limit the selection. Quite tricky to achieve that check..

As for telemetry, it is not possible to limit the range and not possible to have any info about it.  The `lcd.drawSwitch(252)` will actually draw a useful name (like `RxBt` for example), but in code we have no idea about it. We just have a number 252, we don't know the name of it (the user knows, but we don't!) and we don't know if this source exists at all. We can use `getValue()` to check but if it returns 0, we don't know if such counter exists but is not available at the moment or if it's invalid.

If we could invoke `getFieldInfo(252)`, we could get info if it's a valid source and what is its name.

There are some things in the code like `includeInputs`, `includeLogicalSwitches` - these are booleans and they decide if such things should be allowed to be selected by the user. You can assume they are all true for simplicity.

The `n` parameter is the source number / ID, and is used also by `lcd.drawSwitch(n)` and by `getValue(n)` later on.

```lua
InlineSource._maxSource = 249 + 3 * 32 -- assuming 32 tele counters

--[[ valid source numbers

  1 .. 32 inputs
  87 .. 95 sticks and sliders
  96, 97 none (?)
  98, 99 ex sliders
  100 max
  101 .. 103 CYC
  104 .. 109 trims
  110 .. 117 switches
  118 .. 181 logical sw
  182 .. 197 TRxx
  198 .. 229 CHxx
  230 .. 238 GVx
  239 batt
  240 time
  241 gps
  242 .. 245 [--] (?)
  246..248 timers
  249 ... ??  telemetry
]]

function InlineSource:_isSourceValid(n)
  
  -- inputs
  if n >= 1 and n <= 32 then
    -- inputs
    local input = model.getInput(n - 1, 0)
    return self.includeInputs and input ~= nil
  
  -- sticks and sliders
  elseif n >= 87 and n <= 95 then
    return true
  
  -- none (maybe horuse 12?), ex sliders, max, trims, switches
  elseif n >= 96 and n <= 117 then
    return true 
  
  -- logical switches
  elseif n >= 118 and n <= 181 then
    local ls = model.getLogicalSwitch(n - 118)
    return self.includeLogicalSwitches and ls.func ~= 0
  
  -- trainer
  elseif n >= 182 and n <= 197 then
    return self.includeTrainer
  
  -- output chanel CHxx
  elseif n >= 198 and n <= 229 then
    local ch = model.getOutput(n - 198)
    return self.includeOutputs and ch ~= nil
  
  -- GVx - global variables
  elseif n >= 230 and n <= 238 then
    return self.includeGlobalVariables
  
  -- battery, time, gps, timers, weird stuff
  elseif n >= 239 and n <= 248 then
    return true
  
  -- telemetry
  elseif n >= 249 and n <= InlineSource._maxSource then
    return self.includeTelemetry -- telemetry, but we have no way to figure out if such entry is used..
  end
  
  return false
end
```

azaz44, 2019-02-15:
One update: there are also sources which are outputs of mixing scripts. I have no idea how these could be used, are they usable in LUA at all? I mean if you know the name, you can `getValue(name)`, but how to find them out, enumerate, let user select them?

3djc, 2019-02-15:
Looks like telem1-telem32 might not work as intented, we will have a look at that

lshems, 2019-02-15:
Be aware these magic numbers are different between radios and opentx version s.

azaz44, 2019-02-15:
> Be aware these magic numbers are different between radios and opentx version s.

Ok, but is there any way to make it better in LUA?

Some of them could be get by names, but for telemetry it's no option at all.
Switches also problematic, because each radio might have any number of them.
I could possibly try something like `getFieldInfo("SA").id` and `getFieldInfo("LS01")` and assume switches are in between, but this sounds hacky too.

azaz44, 2019-02-15:
> Looks like telem1-telem32 might not work as intented, we will have a look at that

and at mixingSriptOutput1-mixingSriptOutput64 ;)

PR #6335 (2019-02-18): Add telem1-32 and lua mixer outputs to lua fields
This fixes #6241
closed 2019-02-18 by bsongis