opentx/opentx #6245: X10 Wired Trainer Master not working, FrOS it works

Adk90, 2018-11-14:
I'm struggling with wired Trainer Setup on X10S running OpenTX 2.2.2

Did the following test.
1. Connection X7 (Student) to X10S (Master) is not working.
2. Same setup, pluging cable into X12S as Master is working.
3. Changing roles X10S as Student and X7 as Master is working.
4. Did Flash X10S with FrOS and it works flawless with X10S as Master and X7 as Student.

Looks like X10S with OpenTX is not getting the PPM data stream from Jack.
Looking on the circuit board, there is the tip of plug (Left) connected direct, while middle (Right) has a capacitor towards ground. Mass from Plug got some 0 Ohm SMD unit to ground. I did also check with some Mono cable and suggestion to connect middle to ground like mono with no success. 

While playing around with X10S on OpenTX, did get some Student signal by plugging the jack not fully in, like 1mm short. But by just touching the plug or cable I got signal loss message. Looks like wrong pins are used in jack. I'm sure, if it works on FrOS it should work on OpenTX as well.  Let me know if you need more investigation.

3djc, 2018-11-16:
There could be an issue with OpenTx here, but that sounds rather unlikely that you are the first one to notice after thousands and thousands firmware downloads.

What could happen is that your plugf is deffective and FrOS doesn't read the switch state and just looks for signal (not saying it is the case, I do not know, but that could explain it).

Unfortunatly, I dont think we have an x10 to be able to replicate that, so another x10 owner could confirm if it works or not for him, that would be helpfull

Adk90, 2018-11-16:
People do struggle with wired body cable.
Happy if someone having challenges as well could also share his experience.

I use Buddy Diagnostics in Systemmenu to check. To what switch do you refer to, the build in jack switch?
I did had a closer look into the circuits. On X7 PCB looks to me like the switch inside the 3.5mm Jack is used and wired. DT5 top / DT2 and C1 Switch Side, opens when connected[image]. 

On X12S there is only tip and GND used, but there is kind of inverter or Level adjustment circuit with Q1 on the Back side of the PCB. 
[image]
[image]

While on X10S there is the top direct connected to the socked (pin 8) . Additional on the middle there is a little C towards ground and connected to (pin 7) and GND to (pin 6) on 11pin socket assuming yellow as Pin1.
[image]

I did some more testing by using an open 3.5mm plug, where I'm able to just connect wire by wire. When I did top pin without GND. I did got sometimes the student signal (on oscilloscope signal looked floating with little PPM spikes), as soon as Ground got connected as well, Student signal is lost. Kind of describes what I did get with not fully inserted plug.
Once fully connected, I got a nice clean PPM Signal on top jack pin and Pin 8 on 11 Pin Socked (P2).
¯¯¯¯|¯|¯|¯|¯|¯|¯|¯|¯|¯¯¯¯¯¯¯¯¯¯|¯|¯|¯|¯|¯|¯|¯|¯|¯¯¯¯¯¯¯|¯|¯|¯|¯|¯|¯|¯|¯|¯¯¯¯¯¯¯|¯|¯|¯|¯|¯|¯|¯|¯|¯
 So to me it really looks like kind of wrong PIN, flank or signal source is used. As signal can't get better on PCB and jack level.

Adk90, 2018-11-16:
Let me know how I can support you the best. Also happy to beta if required. 
FW seetings (de / eu / noheli / lua).

3djc, 2018-11-21:
Ok, sorry it took us so long to answer this one, we have been working with FrSky on this. Your issue is actually because you try between an x7 and an x10. Had you try with an x9, you would have not seen an issue.

There is a hardware discrepency between x10 and x7, that do not use the same pin to detect.

The fastest workarround is to use a two segment plug, that will in essence link the two detect pins and make it work.

There is also a hardware fix that is possible, linking pin 2 and 4 of the jack plug on x10 as shown in yellow bellow (Note : this is NOT recommended unless you agree to void your waranty and are at ease with soldering tools and techniques)
[image]

Finaly, we will investigate if a software fix is possible by doing a pin less dectection, but that will take some time

Adk90, 2018-11-23:
Thanks a lot.
X7 to X10 with stereo (3pin) and mono (2pin) is now working. 
On top also feeding CPPM signal from R-XSR receiver in into trainer port is working, which allows me to have a true wireless trainer link from any TX capable to bind with the R-XSR receiver.

Details on this Wireless Trainer Link:
I use GND and Switched battery pins in Module bay to feed an LM7805 Voltage regulator to voltage protect and source R-XSR. The CCPM output of R-XSR  is than connected with a 1k resistor (just used Resistor to protect pin) to the top pin on plug. Ground goes to ground lowest pin. 
If a stereo plug is used, just connect middle and ground together. Well it even works without ground on the plug, but if stereo is used, middle pin and low have to be shorted, like a mono plug. 
By turning on PPM on external Module, the R-XSR gets powered on. Having the plug in Trainer Port, I do get student signal from TX bound to R-XSR.

Adelmos, 2019-03-20:
Same issue with X12S (OpenTX 2.2.3) as Master and X9D (OpenTX 2.1.9) as student. If I flash X12S with FrOS, it works flawless as Master and X9D as student. No problem with inverted rules and both with OTX

moelski, 2019-07-31:
Hi !
Are there any fixes on this issue available? I habe the same problem using X10S as Master and X7 as slave. 
And I tested you soldering solution. This did not work on my setup. Will try the stereo plug later (just have to buy one). 

So it would be very welcome if this bug could be fixed in the near future. 
This is the second time where my X10S was a complete fail. There is also no SBUS input in the module bay for connecting an external receiver directly. My X7 did it very well ... But thats another story ...

3djc, 2019-07-31:
Well, a fix if we can make it, will be no different in essence from the soldering patch, so if it does not work for you, there must be another issue somewhere

SnappyFPV, 2019-08-07:
Use a Mono Cable works fine with X10S as Master and X-Lite S as Slave. Pin is Signal, Base is GND
[image]

moelski, 2019-08-18:
Hi !
So just did some Tests. My setup is working now - even if I don´t understand the problem in common ... 
I have to use a stereo pluf on my X10 where the signal is at the tip. The middle and the back from the stereo jack must be connected to the ground. 

On the X7 side it´s easier. Just Signal on the tip and ground on the long other end. 

[image]

I just created a small picture just for reference and clearance. 

Dominik

closed 2020-05-15 by stale[bot]