opentx/opentx #6246: FlySky Telemetry for VOLTS type is actually signed

CapnBry, 2018-11-16:
**Describe the bug**
I have connected FlySky receivers through an iRangeX 4-in-1 module. Normally the values look good (RX voltage, RSSI, etc) but when I added telemetry from iNAV, negative values show as their twos-compliment unsigned versions. "Altitude" from iNAV which is calibrated to 0m on startup shows 655.3m if the model is lowered slightly. "GPS Lat MSB" and "GPS Lon MSB" show similarly wonky numbers for South latitudes and West longitudes, respectively (e.g. 572.72 for -82.64).

Standard AFHDS2A protocol transmits all values as 2 raw bytes and a stock FS-i6 transmitter shows voltage type values properly as signed. I traced it back through the whole chain and looks like the raw data is just forwarded from the multiprotocol module for interpretation in opentx. I believe that all VOLTS type values are actually 16 bit signed, and all RPM type values are 16 bit unsigned.

I should reiterate that this isn't iNAV specific, stock FlySky transmitters properly show negative telemetry values for voltage telemetry types. The multiprotocol module doesn't interpret the data either, it just forwards it, so fixing the interpretation is in opentx's court.

The fix is simple, to just cast the int32 value as int16 without altering the bits.
https://github.com/opentx/opentx/blob/2.2/radio/src/telemetry/flysky_ibus.cpp#L106
~~~
diff --git a/radio/src/telemetry/flysky_ibus.cpp b/radio/src/telemetry/flysky_ibus.cpp
index 7f54c6594..980b158ce 100644
--- a/radio/src/telemetry/flysky_ibus.cpp
+++ b/radio/src/telemetry/flysky_ibus.cpp
@@ -103,6 +103,9 @@ static void processFlySkySensor(const uint8_t *packet)
       else if (id == FS_ID_TEMP)
         // Temperature sensors have 40 degree offset
         value -= 400;
+      else if (sensor->unit == UNIT_VOLTS)
+        // Voltage types are signed 16bit integers
+        value = (int16_t)value;
       setTelemetryValue(TELEM_PROTO_FLYSKY_IBUS, id, 0, instance, value, sensor->unit, sensor->precision);
       return;
     }
~~~

Tested on the hardware listed below as well as a simple C++ test app on x64. A1 (RX Voltage) and A3 (Ext Voltage) read properly with the old code and new, as they are positive. Altitude, Current, and GPS Lat/Lon MSB display proper negative values in the OpenTX telemetry screen with the new code.

**General information**
- OpenTX version 2.2.2 (and in git master/HEAD)
- FlySky Taranis Q X7
- iRangeX 4-in-1 multiprotocol module
- FlySky FS-iA6B receiver and Flit 10+ (FlySky IBUS compatible) receiver
- iNAV 2.0

qba667, 2018-11-16:
@CapnBry  are you sure that my implementation in I6 is handling it all correctly?
I don't know what is the current status of implementation in INAV because I just made it for Betflight/cleanflight. However I am waiting for decent implementation of telemetry in NV14 (hardware) then I will rework it in OpenTX.

qba667, 2018-11-20:
@CapnBry author of this implementation miss the concept a little please check this one:
https://github.com/betaflight/betaflight/blob/master/src/main/telemetry/ibus_shared.c
I will check it together with NV14 implementation - maybe then I will improve implementation in iNAV.

CapnBry, 2018-11-20:
Receiver telemetry implementation aside, negative values coming in voltage type fields need to be fixed on the opentx side to match the FlySky display. It doesn't matter what values are being sent or how they are broken apart, the issue remains that a negative voltage value is supported and displayed properly in stock FlySky transmitter firmware, and opentx displays them as unsigned.

The telemetry code you linked only sends GPS data if built with "IBUS Extended Telemetry" which is just 32bit telemetry data instead of 16 bit. Stock FlySky firmware does not support this, it is only available on the mod firmware. iNAV I believe can support this as well already but this is irrelevant to this conversation, which is the proper display of 16 bit signed values in opentx from IBUS telemetry.

schwabe, 2018-11-20:
I am fine with this change if Flysky remote actually display negative values. If we have more unsigned vs signed types, we might want to add a field to the sensors struct like spektrum telemetry has.

Can you submit this change as pull request?

CapnBry, 2018-11-26:
@schwabe I've posted a pull request for this change, let me know if there's anything else you need from me. It appears there are only 3 data types apart from the RSSI/LQI information: temperature, 16-bit signed, and 32-bit unsigned which is float x 100. The other two types display properly already, matching what is on the FlySky original firmware.

schwabe, 2018-11-26:
@CapnBry it is not float. Float is "floating point". And here the point is not floating. It is fixed point as the point is always at the same point :)

kilrah, 2019-01-06:
Done in #6252

closed 2019-01-06 by kilrah