opentx/opentx #6250: help to read pins stm32

mikerc2017, 2018-11-18:
hello developers 
Can help me read pins on STM32F205VET6  from source ? i do not understand .
EX :read pins from X7 .

kilrah, 2018-11-18:
???

mikerc2017, 2018-11-18:
> ???

yes. i mean diagram connect on X7 taranis . Pins connect to SD CARD, LCD , button , switch .......from stm32f202vet6
I saw source code from target/taranis  but not yet understand Gpio ....  Can you help me a example ?

thanks

mikerc2017, 2018-11-18:
#include "opentx.h"

#define LCD_CONTRAST_OFFSET            160
#define RESET_WAIT_DELAY_MS            300 // Wait time after LCD reset before first command
#define WAIT_FOR_DMA_END()             do { } while (lcd_busy)

#define LCD_NCS_HIGH()                 LCD_NCS_GPIO->BSRRL = LCD_NCS_GPIO_PIN
#define LCD_NCS_LOW()                  LCD_NCS_GPIO->BSRRH = LCD_NCS_GPIO_PIN

#define LCD_A0_HIGH()                  LCD_SPI_GPIO->BSRRL = LCD_A0_GPIO_PIN
#define LCD_A0_LOW()                   LCD_SPI_GPIO->BSRRH = LCD_A0_GPIO_PIN

#define LCD_RST_HIGH()                 LCD_RST_GPIO->BSRRL = LCD_RST_GPIO_PIN
#define LCD_RST_LOW()                  LCD_RST_GPIO->BSRRH = LCD_RST_GPIO_PIN

bool lcdInitFinished = false;
void lcdInitFinish();

i si not clearly same ER9X or 2560 .

kilrah, 2018-11-18:
Sorry but I have no idea what you want to know.

mikerc2017, 2018-11-18:
> Xin lỗi nhưng tôi không có ý tưởng những gì bạn muốn biết và với mục đích gì.

> Sorry but I have no idea what you want to know and to what purpose.

schematic QX7

3djc, 2018-11-18:
Radio schematics are not open source, and such requests should be addressed to FrSky

closed 2018-11-18 by 3djc

kilrah, 2018-11-18:
Look into hal.h.