opentx/opentx #6251: Companion should refuse to write firmware if "taranis" drive is not mounted "sync"

prof7bit, 2018-11-19:
Problem observed with current 2.2.2 release and Q-X7 in bootloader mode.
PC is (K)Ubuntu Bionic, the current LTS release of (K)Ubuntu.

I followed the instructions about turning on my Taranis in Bootloader mode (trim switches inward) and the plugged in the USB. Two removable drives were detected and automatically mounted.

Companion was then able (or rather pretended to be) to write the latest firmware to the Taranis. After it completed without error I issued a "sync" in the terminal (just for good measure as I always do), then safely ejected both drives, pulled the plug, choose "exit" on the Taranis bootloader menu and **the screen went black**! I could not turn it on again! I already thought I had bricked it but then I found the bootloader still worked.

I tried one more time with a differently configured firmware and this time it came just beyond the switch warning and then was stuck in a boot loop.

This reminded me of a similar problem a year ago that occurred to me with a PEmicro OpenSDA JTAG probe that also was supposed to be upgradeable via mass storage file write and that also worked only with Windows and not with Linux, this was a well known problem with this particular device and the reason was the bootloader on that dongle had a **broken mass storage implementation** that only happened to work by sheer coincidence when windows was used to write the file and failed in subtle ways on any other OS. 

After remembering this I booted my Windows PC, installed companion and could write working firmware images **using my windows PC without problems**! 

When looking in my mtab I can observe that the automounter by default mounts the vfat device with mount option "flush" which means it is still cached but cache is flushed earlier. This is generally optimal for removeable flash drives but its still not sync!

I added an entry to my fstab "LABEL=Taranis" with mount option "sync" to override the default mount options and plugged it in again and confirmed this time it was mounted sync. I attempted a few firmware writes with companion and each of them succeeded now! The progress bar also took noticeably longer now.

--
Conclusion:

Option 1: Fix the bootloader so it will be able to deal with the blocks written in arbitrary order when the write cache is flushed after the file write, probably quite hard to implement, maybe not enough RAM?

Option 2: Detect the mount options of the Taranis volume and refuse to write to it (or at least refuse to do things that would be messed up) if it is mounted with write caching enabled.

Option 3: **At least** add some big red warnings in the docs and maybe a small howto for popular non-windows OS on how to disable the write cache for the taranis volume.

schwabe, 2018-11-19:
Windows always mounts drives like this these without cache. 

For the bootloader to actually screw up writing, Linux must write in a pretty random order to the drive.

prof7bit, 2018-11-19:
Well, obviously the write pattern that occurs when the cache is flushed can somehow trigger unintended things to happen and ends up with messed up firmware on the device. If your comment means that it should not happen then it is obviously a bug in the bootloader.

MikeBland, 2018-11-22:
The 'problem' is likely to be caused by the flash being organised in sectors (16K) that need to be erased.
Currently, a sector is erased when the address to be flashed matches the first address of the sector. This works fine for sequential writing. Clearly, is the flash is written in a 'random' order, this could cause data already written to be erased.
The bootloader could be changed to erase a sector on the "first" write to any address in the sector, then note that the sector has been erased so not erase it again, but then, if you go to flash the firmware again, the bootloader would need to detect the erasing would need to be done again.

raphaelcoeffic, 2019-08-14:
@schwabe @MikeBland is that actually something that we could fix? or should we just document what is explained here?

prof7bit, 2019-08-14:
> is that actually something that we could fix? or should we just document what is explained here?

It can easily detected whether the drive is not mounted "sync" and then instead of writing the firmware it could pop up a message box explaining the problem and a link explaining possible solutions. 

There are probably many solutions, on my system I just added a line to my fstab:

````
LABEL=Taranis                               /media/bernd/Taranis    vfat    rw,nosuid,nodev,noauto,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,sync,errors=remount-ro 0 0
````

Now whenever I plug it in it will be mounted sync and then everything works like a charm.

closed 2020-06-06 by stale[bot]