opentx/opentx #6257: Companion support for Horus telemetry screens

Luznatural, 2018-11-29:
Companion 2.2.2 has a poor support any management of the different existing options available on Horus X10 and X12S telemetry screens.
When you edit a model using companion for Horus X10/X10S, or Horus X12S, you cannot manage the telemetry screens: add or remove screens, top bar options, add or remove widgets, nor change widgets options, under the telemetry tab.
The only way to do this is making it manually directly on the radio.
I Expect behavior to have the same options available if both radios with Open Tx.

**General information**
 - OpenTX version 2.2.2
 - Hardware: Frsky X10/X10S and Frsky X12S
 - OpenTX Companion 2.2.2 (Windows version)

lshems, 2018-11-29:
You can do that with the simulator. Start the simulator and choose the correct settings.

Works for all radio types.

You can evendiscover sensors I you set some simulator vaues for telemetry.

Luznatural, 2018-11-30:
Thanks for the información.
I will test it this afternoon, as soon as I come back from work.
But this way is only a mitigation measure, as all options should also be available in Companion.
I did not even thought to use the simulator to change settings in a model.

lshems, 2018-11-30:
I think they are working on integrating the simulator and companion, but if you want to be able to set up widgets without a simulated screen, how do you envision that? 

Isnt your prime request that you can want a seamless integration of companion, simulator and radio?

Just trying to clarify your needs

Luznatural, 2018-12-06:
Sorry for the delay. I have been extremely busy with my work, with the closure of the year.
I fully reckognize that X10 and X12 radios are more expensive than other tx's, and as a consequence less popular that Taranis +. But this should not be an impediment to evolve a little bit the software. If you ask anyone which radio he/she would like to have if money wasn't an issue, no doubt everybody would select Horus, as they are far more sexier than Taranis.
In my case, I received the X10S as an anniversary gift from my wife (of course, I was delighted). So, I have almost forgotten my old Taranis, except for doing things in the field that you already cannot do with the Horus (for instance updating firmware of a receiver with the Tx, or exchanging plane configuration to other friend which have the same plane). Since I took my Horus to the field, three more friends also bought X10S in a couple of weeks.
Coming back to the issue, I feel Horus Tx's should receive a little more priority from the developong team. I do not see the same speed of opentx evolution for Horus than happens in other later radios.
In my humble perspective, I do not see practical the effort of linking simulator with companion, before further developing companion to deliver better support to Horus.
I am a plane flier and personally don't have an urgent need to use simulator. Of course I have tried it, but I normally know what I want to do with the radio, as the options are normally limited in the setup, mixers, flight modes, global values, or special functions.
All regular simulator users I know are normally multicopter fliers, which are far more complex to set than planes, but only when they do their initial setup. As when they set the drone, they don't tend to come back to the simulator. So I guess they have increased needs, but not too many.
For me, companion should provide at least the same options available in the respective radio you use. In the case of Horus (doesn't matter both X10/X10S or X12S) we are talking most of the time in modify companion for being able to use check boxes and providing down lists with the options available that you could check to display.
As an example, if you want to put the batt value in the top bar, you should have the option of selecting the edittion of top bar, select the box where you want to display batt value in the top bar, and select batt. Then, you could select the next box, and select Rxbatt (as example) and so on. Finally, save the options and continue configuring the rest.
Same thing could be done with the widget configuration, select the desired option, etc, in all telemetry pages, and in general in all menus of the radio.
I am not a programmer, but I will gladly contribute in the development companion for Horus.
Sorry for the length of the response.
Happy flying

3djc, 2018-12-06:
That would be a massive development work for very little benefit, but if you  find somebody to do the job, we will look at the PR.

As for Horus getting less support, it is untrue, and by the way, Companion does not allow either to setup Taranis telem screen in a way similar to what you described, despite having been there much much longer

kilrah, 2018-12-06:
Everything model related is "set in stone" aka it is known in advance what the choices are, how many of a given item can be present etc.

Horus UI configuration is completely dynamic which makes it super flexible, but it is unknown what is possible until runtime since layouts and widgets are loaded from whatever the user has added on the SD card. For companion to know what settings and optins to give you it would need to know exactly what firmware you have on the radio, have a copy of the SD card, AND be able to read, understand and execute lua widget scripts so as to know what they do, what options they offer etc so you can set them up. 
That's really complicated and nobody really found the courage to try and make it happen. Due to the same reason lua mixer scripts cannot be configured in companion for any radio.

kilrah, 2018-12-06:
The whole point of companion is to be able to select and enter things using the greater comfort of a computer with a big screen, a mouse and a keyboard to select things without scrolling and to actually type numbers. 

It's perfectly normal to want that for the screens as well. Would for sure be more convenient to just have a page with a dropdown where you select the layout, said layout zones appear with their own dropdowns to select the widget, once selected that brings up the options for that widget. You could set things up in 4 direct clicks instead of entering, scrolling, adjusting, exiting repeatedly many times. 

People don't use companion "just for the sake of setting things up on a computer", it's becasue of the added comfort. The simulator brings nothing at all in that regard, if anything it's probably easier on the radio itself due to the instant feedback and better controls.

lshems, 2018-12-06:
You just gave the requirements I was referring to:

Greater screen layout,better support for mouse and stuff.

That could relatively easily be achieved. I do it with screen zoom already. 

Yes, buttons are tedious but could really easily be fixed, if given some thought.

???!!!

kilrah, 2018-12-06:
That's not what people are looking for at all.

lshems, 2018-12-06:
That's what you think people want, as seems the initiator.

I'm people too, and want something more achievable.

But hey, don't bother. I was just trying to point people at what can be done already.

lshems, 2018-12-06:
On top of that, for model logic your example makes sense.

For widget layout to be used on the radio screen, not at all. You would need a mockup of the screen anyway, as in the simulator.

So I think you're oversimplifying your approach.

Again, it's o so constructive this place.

Creativity just soaks out of every corner.

kilrah, 2018-12-06:
We've had many requests about that feature and they all went in the direction I'm mentioning.

> I was just trying to point people at what can be done already.

They told you twice already that it's not what they're looking for. 

> You would need a mockup of the screen anyway

I described exactly that in my previous post, but as usual you don't care, or completely fail to understand how things work, the technical differences between the 2 things and why they're there in the first place.

lshems, 2018-12-06:
I'm not ditching anything. That's your words. 

But ok, I'll admit, I was overreacting here. My fault, and apologies.

The requirements, after reading them three times over, are pretty clear. But, as also mentioned, not that close to being a development easily done.

The Horus screen setup on the radio (or simulator :) ) is not that awkward, and with a few tweaks rather effective.

I still think a little revamping of the simulator and a more flexible integration with  companion might be a feasable short term solution, with the risk of staying the only solution.

Luznatural, 2019-08-23:
I Continue interested in this issue

closed 2020-05-19 by stale[bot]