opentx/opentx #6259: Model.getcurve() and model.setcurve() use different indexing on points

lshems, 2018-12-02:
Model.getcurve() returns parameter y indexed from 0 up, the model.setcurve() wants them indexed from 1 up.

Please fix now or fix never. You can program around, but please not version dependant.

lshems, 2018-12-03:
Sorry, but almost everything is zero start indexed. So best to adapt new setcurve.

notlebowski, 2018-12-07:
Or better even...
Make a new function model.curve
You would have the compatibility part solved and it is apparent that they can be used for getting and setting the curve.
When I created my first script [https://www.justfly.solutions/index.php/full-repository?view=product&layout=plugin&format=raw&field_id=69&id=28] I found out that the output of getCurve could not be used for setCurve   ?????

lshems, 2018-12-07:
I suggested multiple times to get rid of the set and get. On function is enough.

Local C={}

Loop something
   C=model.curve(1,C)
   Modify C ....
End loop

Would simplify a lot of scripts where the modify would depend on some userinteraction.

schwabe, 2018-12-07:
Nice idea but does not work well with the way lua is designed.

lshems, 2018-12-08:
Please explain?

I created a library with exactly this way of using curve(), using the getcurve and setcurve, but for all unnecessary duplicate pairs with get and set.

Makes my code much easier to read and maintain.

So what within Lua doesn't work here?

lshems, 2018-12-08:
Just fix the stupid bug, or decide to leave it there forever now. It's you guys that keep breaking scripts.

I never asked, only suggested.

But hey, I'll work around again.

notlebowski, 2018-12-08:
It may have been just my "luck" that the first time I tried LUA I was immediately confronted with a 2 pair-function that were in fact not exactly paired.
Is it really the only function in the whole LUA-library that isn't properly paired???
I wouldn't know as it immediately put me off using LUA further.

I have no idea how that imbalance started, but I do know that an imbalance wouldn't have been created if it was 1 function that would be able to both set and get as it is then even more obvious that the 2 should stay balanced.

As you don't want to break compatibility a new function could be created. You could start with 1 that could do both get as set. It's not that lshems invented that.

lshems, 2018-12-08:
Please FIX the BUG, instead of trying to keep compatibility for the BUG.

bad practice.

Or do nothing, just promise to never fix it.

notlebowski, 2018-12-10:
Can this be reopened or at least a remark from a dev if something is done about it or not?

lshems, 2018-12-10:
I don't care anymore, seen the type of communication.

See https://www.rcgroups.com/forums/showthread.php?3176435-Bugs-in-openTX-lua-API-s

schwabe, 2018-12-10:
@notlebowski lshem closed the bug. Also @lshems your attitude and now with public shaming makes dev like me less and less likely to care about your problem. 

setCurve was developed for using it in the setup wizard and there reading/store was not the goal, so I choose to use something that was a good fit for lua programming.

I think adding a getCurve_ex/getCurve_new or something similar or a API breaking change is a good thing to do. Unfortunately,  at the moment I have little time for OpenTX, so the best I can do is review a pull request if someone here proposes a change for the better.

@lshems maybe you can focus your energy into doing productive changes and makeing a pull request for this one instead using that energy for complaining.

lshems, 2018-12-10:
You are playing the victim here.

I just posted a bug, with a suggestion on how to improve things, in the context of the bug.

Then you developers choose to post objections, that after inquiry and argumentation from my side, are never made substantial.

I don't want you to do anything, apart from being respectful.

I don't start a thread on my blog to put you on public shame. You are doing that yourselves constantly, no need for me.

I just open a thread to keep track of all the inconsistencies and bugs, for others not to loose precious time stumbling over them. I don't need any of them fixed, so when you start putting words in my mouth and clearly show you are not happy with my requests, well, I close them.

Do as you wish, but stop playing the victim. You're only a victim of your own behaviour.

lshems, 2018-12-10:
Better open a new one perhaps? Because I do not want my request fulfilled anymore. It was a genuine mistake to post a bug, trying to point you at it before it becomes legacy.

I'll never do it again.

schwabe, 2018-12-11:
@lshems please stop being destructive

lshems, 2018-12-11:
Please log a new one. I don't want to be associated with any further things on your actions, or lack thereof.

I'll close again if I feel like it, just like you do on requests from others.

closed 2019-11-15 by stale[bot]