opentx/opentx #6261: Allow R9 Telemetry While Using Internal RF Module For Redundancy

MrLAtt, 2018-12-04:
I'm trying to setup my r9 slim+ with an xm+ for redundancy, but when I enable the internal module for the xm+ I'm no longer able to receive telemetry from the r9 Slim+. (open tx 2.2.2 with latest eu (non-Flex) firmware

Can we instead have the option to decide weather we want telemetry from the internal module or the r9m module, or is there some reason why that can't work?

Its very frustrating since the xm+ does not have telemetry, and even if it did, the telemetry would not have enough range

3djc, 2018-12-04:
Thé is no option to turn off the internal module telemetry, and given the two share the same wire, it can only be one or the other

closed 2018-12-04 by 3djc

MrLAtt, 2018-12-04:
From what I understand, cant we have both active through the same wire, just like you would with s-port on the receiver with multiple telemetry modules, as long as there is not too much info being sent at one time. Which, if using a non-telemetry capable receiver with the internal module there should be no problem?

kilrah, 2018-12-04:
No, amount is irrelevant. Modules are bus masters and poll the bus every 12ms.There simply can't be 2 of them.

MrLAtt, 2018-12-04:
Would this same situation work if the s-port line of the internal module was phisically disconected, since it looks like quite a simple hardware hack to either just cut the trace to the pin on the xjt riser or route it through a switch, or would there be some other software problem stopping it. 

Because the only other solution to this I can think of for my situation (where I need frequency redundancy and telemetry from the 900mhz system) is to have a second transmitter connected to my x9d through the trainer port, and then use that as the transmitter for the 2.4ghz system.

Is there any other solution?

3djc, 2018-12-05:
There are upcomming changes that will further prevent concurent use of iXJT and R9M, so there is no point in going down that path

kilrah, 2018-12-05:
It would work but you'll have to do the required code changes to remove the logic that prevents selection of normally incompatible combinations.

MrLAtt, 2018-12-05:
ok, thanks for the help