opentx/opentx #6265: replace Orginal Frsky by Multiprotocol

jonathan2018, 2018-12-15:
hi Deverloper 
I have problem with Orginal FRSKY  x9 , it short range and now i want to replace by Multiprotocol with Frsky protocol . 
I can not use  space on X9 for modul multiprotocol because i used it for R9M. 
i want to connect direct Multi to orginal Frsky to replace . It can compatible ? or any solution better ?

jonathan2018, 2018-12-16:
> No, multiprotocol is not supported on the internal port.

even do not need telemetry. Internal port use PPM input ?

kilrah, 2018-12-16:
Smart port is common to internal and external. 
We only send PXX to internal and nothing else (actually there can be PPM id you select the "internalppm" option. But multi serial is not available, there is no code written for it to be output by the internal module timer.

jonathan2018, 2018-12-16:
Ohh. i do not see internalPPM in option Companion 2.2.2 ???

[image]

kilrah, 2018-12-16:
I have it here. Order is different.

jonathan2018, 2018-12-16:
> I have it here. Order is different.

[image]

kilrah, 2018-12-16:
It's only for X9. X7 hardware is different.

schwabe, 2018-12-17:
Multi on internal module is possible but so far, no one has made a modification to his/her hardware and coded and submitted changes. As long as this does not change, multi in internal module is not supported.

jonathan2018, 2018-12-27:
> Multi on internal module is possible but so far, no one has made a modification to his/her hardware and coded and submitted changes. As long as this does not change, multi in internal module is not supported.

Even change PID  by modul Multiprotocol a long time nobody fix. maybe it not important with Opentx

schwabe, 2018-12-28:
@jonathan2018 I am not even sure what PID by modul is...

jonathan2018, 2018-12-28:
> @ jonathan2018 Tôi thậm chí không chắc chắn PID của modul là gì ...

sorry for my idiom .A issues from Github multiprotocol that you attended https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/136

schwabe, 2018-12-28:
@jonathan2018 yeah but have you also read my analysis? This as also a HW problem. And the motivation to fix this for the OpenTX platform seem to be relatively low as most people will use the internal module for D16 protocol anyway. On the OpenTX side everything should be fine.

jonathan2018, 2018-12-29:
> @jonathan2018 yeah but have you also read my analysis? This as also a HW problem. And the motivation to fix this for the OpenTX platform seem to be relatively low as most people will use the internal module for D16 protocol anyway. On the OpenTX side everything should be fine.

that is old story. i think everyone will like a multiprotocol now /
Pascan said problem from Opentx . and you said problem from hardware Multi but it worked with Er9xsky ???

schwabe, 2018-12-29:
@jonathan2018 OpenTX does not implement the soft serial for uninverted serial. For inverted serial I am not aware of any new developments.

jonathan2018, 2018-12-30:
@schwabe 
i create a new issues on your branch Multi.   If you have time , talk more there. thanks

jack113113, 2019-02-16:
i saw a report Script display on Qx7 work here https://github.com/midelic/Ersky9x-Tx-bas-scripts/issues/1#issuecomment-464300436
but he used firmware from ersky9x
@schwabe said problem from hardware smartport ????? 
ersky9x work and Opentx can not ?

closed 2019-11-15 by stale[bot]