opentx/opentx #6266: Sport mirror does not send the telemetry of the connected FC

jostak, 2018-12-15:
On taranis x9d, there is no telemetry from the connected FC with inav. Lua script works correctly showing the right information while no information on sport mirror, only the bacic messages regarding RSsi and Rx volt

jostak, 2018-12-15:
> What kind of telemetry are you sending from inav? What are you using to
> read from the serial port?

Hi,

Thank you for replying,

Inav is set for sport telemetry and sends the standard messages to the
receiver sport regarding gps, battery, current, altitude, attitude, plus
encoded flightodes etc.  The sport mirror in taranis battery compartment
monitored with saleae. Seems like only the messages with swr, rssi and Rx
bat are passed onto the sport mirror (means the one generated by the
receiver with ID98), but not all those messages that inav puts into
receiver sport, while it is available for lua sript in taranis and it
works. Need to log later what exactly inav sends to the receiver sport pin
and what are the ID's of these messages, but since these messages are
available for lua sript giving the clear and understandable informations,
why those messages are not put onto the sport mirror. This is what I do not
understand.

Anyway, opentx rules and the only reason of my ticket is that I am working
on the antenna tracker and wanted to use the GPS position reported by inav
via sport telemetry to taranis, get this information out of taranis using
sport mirror and based on this calculate the coordinates for the antenna
pointing. If you know the better way to get this info out of taranis,
please help.

I would like to avoid the parallel link to get this info from the model.

Thank you and regards

kilrah, 2018-12-15:
There are 2 types of telemetry, the standard smart port telemetry that gets parsed and displayed by the telemetry system in OpenTX, and the "passthrough" system that goes straight to lua for decoding. I believe s-port mirror relays only the former.

jostak, 2018-12-15:
Is is possible to get this available as well?

kilrah, 2018-12-15:
Maybe, but needs discussion.

jostak, 2018-12-15:
Ok. Thank you for help and explanation.

Regards

jostak, 2018-12-15:
Hi,

just final thoughts after loging the FC<>XSR communication,

it seems that there is even more "tricks" that I am not aware of...and
which are quite poorly described anywhere:

based on the log, XSR by default pulls for all 28 phisical ID's on SPORT,
while FC with INAV is disconnected from XSR, but after connecting FC to
XSR, XSR seems to change the way it pulls for the data,
it pulls ID28 (0x1B) each second pull and the FC responds for this pull by
giving all those informations, pull by pull with the additional ID's in
accordance to this

// FrSky new DATA IDs (2 bytes)
#define ALT_FIRST_ID              0x0100
#define ALT_LAST_ID               0x010f
#define VARIO_FIRST_ID            0x0110
#define VARIO_LAST_ID             0x011f
#define CURR_FIRST_ID             0x0200
#define CURR_LAST_ID              0x020f
#define VFAS_FIRST_ID             0x0210
#define VFAS_LAST_ID              0x021f
#define CELLS_FIRST_ID            0x0300
#define CELLS_LAST_ID             0x030f
#define T1_FIRST_ID               0x0400
#define T1_LAST_ID                0x040f
#define T2_FIRST_ID               0x0410
#define T2_LAST_ID                0x041f
#define RPM_FIRST_ID              0x0500
#define RPM_LAST_ID               0x050f
#define FUEL_FIRST_ID             0x0600
#define FUEL_LAST_ID              0x060f
#define ACCX_FIRST_ID             0x0700
#define ACCX_LAST_ID              0x070f
#define ACCY_FIRST_ID             0x0710
#define ACCY_LAST_ID              0x071f
#define ACCZ_FIRST_ID             0x0720
#define ACCZ_LAST_ID              0x072f
#define GPS_LONG_LATI_FIRST_ID    0x0800
#define GPS_LONG_LATI_LAST_ID     0x080f
#define GPS_ALT_FIRST_ID          0x0820
#define GPS_ALT_LAST_ID           0x082f
#define GPS_SPEED_FIRST_ID        0x0830
#define GPS_SPEED_LAST_ID         0x083f
#define GPS_COURS_FIRST_ID        0x0840
#define GPS_COURS_LAST_ID         0x084f
#define GPS_TIME_DATE_FIRST_ID    0x0850
#define GPS_TIME_DATE_LAST_ID     0x085f
#define A3_FIRST_ID               0x0900
#define A3_LAST_ID                0x090f
#define A4_FIRST_ID               0x0910
#define A4_LAST_ID                0x091f
#define AIR_SPEED_FIRST_ID        0x0a00
#define AIR_SPEED_LAST_ID         0x0a0f
#define RBOX_BATT1_FIRST_ID       0x0b00
#define RBOX_BATT1_LAST_ID        0x0b0f
#define RBOX_BATT2_FIRST_ID       0x0b10
#define RBOX_BATT2_LAST_ID        0x0b1f
#define RBOX_STATE_FIRST_ID       0x0b20
#define RBOX_STATE_LAST_ID        0x0b2f
#define RBOX_CNSP_FIRST_ID        0x0b30
#define RBOX_CNSP_LAST_ID         0x0b3f
#define SD1_FIRST_ID              0x0b40
#define SD1_LAST_ID               0x0b4f
#define ESC_POWER_FIRST_ID        0x0b50
#define ESC_POWER_LAST_ID         0x0b5f
#define ESC_RPM_CONS_FIRST_ID     0x0b60
#define ESC_RPM_CONS_LAST_ID      0x0b6f
#define ESC_TEMPERATURE_FIRST_ID  0x0b70
#define ESC_TEMPERATURE_LAST_ID   0x0b7f
#define X8R_FIRST_ID              0x0c20
#define X8R_LAST_ID               0x0c2f
#define S6R_FIRST_ID              0x0c30
#define S6R_LAST_ID               0x0c3f
#define GASSUIT_TEMP_FIRST_ID     0x0d00
#define GASSUIT_TEMP_LAST_ID      0x0d0f
#define GASSUIT_SPEED_FIRST_ID    0x0d10
#define GASSUIT_SPEED_LAST_ID     0x0d1f
#define GASSUIT_FUEL_FIRST_ID     0x0d20
#define GASSUIT_FUEL_LAST_ID      0x0d2f
#define DIY_FIRST_ID              0x5000
#define DIY_LAST_ID               0x52ff
#define DIY_STREAM_FIRST_ID       0x5000
#define DIY_STREAM_LAST_ID        0x50ff
#define FACT_TEST_ID              0xf000
#define RSSI_ID                   0xf101
#define ADC1_ID                   0xf102
#define ADC2_ID                   0xf103
#define SP2UART_A_ID              0xfd00
#define SP2UART_B_ID              0xfd01
#define BATT_ID                   0xf104
#define RAS_ID                    0xf105
#define XJT_VERSION_ID            0xf106
#define FUEL_QTY_FIRST_ID         0x0a10
#define FUEL_QTY_LAST_ID          0x0a1f

anyway, all those responses on pull ID28 (0x1B) does not come onto the
sport mirror in Taranis and it makes my project more demanding.

OK, thank you & greetings from Poland

Merry christmas

kilrah, 2018-12-15:
If those come then you should see sensors when you discover them in the radio.

closed 2020-04-06 by stale[bot]