opentx/opentx #6268: Companion can't find sound files.

Smedly101, 2018-12-16:
Opentx Companion V 2.2.2
SDcard V X7-2.2V0018 with Amber sound pack
Windows 10
In X7 profile, on the special functions page with play track selected, no drop down list of sounds appears and cannot play sound with play arrow. "Unable to find or open sound file" error message appears. Sounds are normal in simulation mode and in X9D profile. Triple checked SDcard folder location and permissions, tried different locations and drives, also tried 2 other computers with fresh install, Windows and Linux...all the same result.
[FirmwareDebug_18-12-15_19-48-32.log](https://github.com/opentx/opentx/files/2683249/FirmwareDebug_18-12-15_19-48-32.log)

3djc, 2018-12-16:
I do not seem to be able to replacate (W10, x7 profile)

[image]

Smedly101, 2018-12-16:
Thanks for the response, the clip that you sent me prompted me to try a new document1.otx and this worked fine, the sounds were available. Previously I was working with an eprom.bin from the X7 transmitter and that still has the issue. I will try to reprogram my models into a fresh document1.otx hopefully that will work.

Smedly101, 2018-12-16:
OK, problem solved, must have been a corrupted EPROM, just needed to transfer models to new document1.otx. Thanks for the help.

closed 2018-12-16 by 3djc