opentx/opentx #6270: taranis x9d+ global backlight keys+timeout no longer working - backlight stays on

pitts-mo, 2018-12-16:
I have recently updated from 2.2.2 to nightly 2.2.3N683, I prefer back-light on keys only for 5 seconds but have found back-light will no longer turn off unless back-light is distinctly set to off.

pitts-mo, 2018-12-16:
This is my first time looking at this code and I am a bit overwhelmed. 

But I currently suspect code change for issue #6145 has hit a need to differentiate power switch types of target hardware ( power slide switch / power push button ).

radio/src/targets/taranis/keys_driver.cpp 
if ((keys_input || trims_input **|| pwrPressed()**) && (g_eeGeneral.backlightMode & e_backlight_mode_keys)) {

pitts-mo, 2018-12-20:
I have confirmed my issue of backlit stuck on is bug related to #6145. I am willing to make and test code changes if someone can suggest the correct fix.

confirmation detail:
new docker and git build about an hour ago

opentx-x9d+-noheli-luac-lua-en-20181219.bin = back light still stays on

Removed "|| pwrPressed()" from line mentioned in prior comment 
opentx-x9d+-noheli-luac-lua-en-20181219-pitts-mo.bin = back light will now turn off after my defined setting of 5 seconds.

opentx-x9d+-noheli-luac-lua-en-20181219.bin = confirmed back light is stuck on for much more than 15 seconds and my setting was verified to be still set to 5 seconds.

3djc, 2018-12-20:
I have done the fix, Thanks for your investigation and report :)

closed 2018-12-20 by 3djc

pitts-mo, 2018-12-20:
Thank you @3djc,
Utilizing a new git clone a few minutes ago I have confirmed the backlight keys only+timeout behavior I desire on x9d+ has returned.

commit 3a48bab4 (2018-12-20): Fix backlight not turning off on radio with physical on/off switch
This fixes #6270
commit 74b9446a (2018-12-20): Fix backlight not turning off on radio with physical on/off switch
This fixes #6270