opentx/opentx #6273: Request: inform that a sensor is missing

notlebowski, 2018-12-22:
OTX informs us when a sensor is not sending any info anymore.
It doesn't inform the user however when a defined sensor is not sending at all.

On some models you don't want to fly if not all sensors are working. It would be nice if OTX would inform us then as well.
Best is that this check is optional for each model and off be default if it gets implemented.

To achieve this on current setups I need to use several logical switches

kilrah, 2018-12-22:
Quick glance at the telemetry screen that everything is there?

notlebowski, 2018-12-23:
Trying to do this with logical switches turns out to be difficult.
I have no way of finding out I'm not getting live data from the sensor.
I thought of testing for Cells > 1V
But testing for >1V also returns true if that telemetry field once received a valid value.

L31 Sticky(!Telemetry, L32) Delay(5s)
L32 Telemetry AND ---- Delay(5s)
L33 Edge(!L31, [0:instant])
L34 ( TELE8:CLow < 4.1 ) AND L31 Delay(2s)
L35 TELE7:C4 > 1
L36 L33 AND !L35

L33 turns momentarily true when telemetry should be working fully after not working for at least 5 seconds.
After detecting a lowest cell voltage lower than 4.1V (L34) it will tell me the battery is not a full one.
That is properly working.
I can't however think of a trick to tell me I'm not getting live data.`

If the system would just tell me a "sensor is not ready"

In case of the FLVSS the sensor can be attached to the receiver, but not to the balance connector.

notlebowski, 2018-12-23:
This is working....
I didn't think of using the telemetry value as a boolean.
It's even 1 line less

**L31** Sticky(!Telemetry, L32) Delay(5s)
**L32** Telemetry AND ---- Delay(5s)
**L33** Edge(!L31, [0:instant])
**L34** ( TELE8:CLow < 4.1 ) AND L31 Delay(2s)
**L36** L33 AND !C4

**SF31** L36 - Play Track (sensnr)
**SF32** L34 - Play Track (nfchrg) played once, not during startup
**SF33** L34 - Play Value (TELE8:CLow) played once, not during startup

notlebowski, 2018-12-23:
I think I now have covered it.
Turning on the receiver before the radio (not good) will spoil it a bit, but the focus is on noticing a battery swap whilst leaving the radio on. It needs to tell me the replaced battery is full or not. An in-between landing with a battery disconnect/connect will not reset the timer or consumption and reminds you of the battery voltage.

**L31** Sticky(!Telemetry, L32) Delay(5s)
**L32** Telemetry AND ---- Delay(5s)
**L33** Edge(!L31, [0:instant])
**L34** ( TELE8:CLow < 4.1 ) AND TELE7
**L35** L33 AND L34
**L36** L33 AND !TELE7

**SF31** L36 - Play Track (sensnr)
**SF32** L35 - Play Track (nfchrg) played once, not during startup
**SF33** L35 - Play Value (TELE8:CLow) played once, not during startup

notlebowski, 2018-12-23:
I have my use case covered, so I will close it.
You may of course reopen it if you think there's merit making this globally.

closed 2018-12-23 by notlebowski