opentx/opentx #6275: possibility of inserting pwm values instead of curving for multi-position keys

manotroll, 2018-12-24:
there is possibility of instead making a curve to try to hit the values allow fixed numbers
such as when using the 6-way switch for flight modes in px4 the program does not allow changing the values

http://ardupilot.org/copter/docs/common-rc-transmitter-flight-mode-configuration.html
or there is a way to do it and I do not know how to do it

3djc, 2018-12-24:
Whats wrong with curve ?

lshems, 2018-12-24:
I'll make you a script to set it up in one click. Check my profile on rcgroups and pm.

manotroll, 2018-12-24:
> 
> 
> Whats wrong with curve ? You can do a LUA that setup your curve the way you want it at model creation
> 
> You can also use mix lines, overides, plenty of ways

But how do I do this?
because by editing the points in the curve comes an hour that one point changes the other
could you give me links that show you how to use the available methods

kilrah, 2018-12-24:
Don't use a curve, use mixer lines with the proper values on the desired channels. 

-100% MAX Switch(6P1)
-80%   MAX Switch(6P2) 

etc...

3djc, 2018-12-24:
Closing this, as this is a training issue rather then an opentx bug

closed 2018-12-24 by 3djc

manotroll, 2018-12-24:
> 
> 
> Don't use a curve, use mixer lines with the proper values on the desired channel.
> 
> -100% MAX Switch(6P1)
> -80% MAX Switch(6P2)
> 
> etc...

How do I do this?

kilrah, 2018-12-24:
[image]

[image]

manotroll, 2018-12-24:
but I'd like to use a single key
because I am using the mod of the rotary switch of 6 possibilities

manotroll, 2018-12-24:
I do not understand
I can not figure out how to get this result