opentx/opentx #6281: OpenTX 2.2.2 download problem

Cattlebaron, 2019-01-02:
I can not download OpenTX 2.2.2, when I click on the link I am taken back to the opening page. I can download the sdcard-taranis-x9-2.2V0018.zip

Cattlebaron, 2019-01-03:
> You do not mention what platfrom, but it seems ok from here.

Thank you for the reply. I use an Apple Mac running the lates operating
system. When i click on the download link it return me to the opening
screen. See the two attachments.

kilrah, 2019-01-03:
You cannot post attachments by mail, need to use the web interface.

We don't know whether you're trying to download companion or firmware, what version, what radio and options if the former, what link you're using if companion...

kilrah, 2019-01-03:
Firmware is downloaded from within companion, not from the website.

closed 2019-01-03 by kilrah

Cattlebaron, 2019-01-03:
Thanks Andre,

I will now have to find out how to do this as previously I would download
the file from the web and then with openTX flash the firmware using the
file I had on my mac, but I will learn and succeed.

Cheers

kilrah, 2019-01-03:
We have never distributed firmware files on the website, so you're probably confusing with something else....