opentx/opentx #6283: X-lite keep disconnected with BetaFPV F4 brushed (matekf411rx) and crash with throttle up

slyers77, 2019-01-02:
my setups are:
board fc : BetaFPV F4 Brushed (betaflight 3.5.4, target Matek F411RX)
X lite opentx 2.2.2 with multiprotocol
binding with D16 and telemetery 9-16

my problem : 
after a while in acro or air mode, suddenly my micro quad is disconnected, sensor loss, and throttle fully up, and nothing i can do, until I re-plug the battery connector. And after several seconds its disconnected again and again
and when in flight, my quad is like wobble...
Is X-Lite openTX not receiver built-in board friendly (yet)?

but when I use my X9D plus TX, no problem

please fix it.... because I like X-LIte very much for its simplicity

thanks

3djc, 2019-01-02:
Given the symptoms you highlight,  there could be a number of issues, from which an OpenTx issue is least likely to be the cause, and there is no known issues with xlite and multi.

So I'm closing this. If after detailled troubleshooting, you have evidence that OpenTx is the source of the issue, feel free to reopen.

closed 2019-01-02 by 3djc

slyers77, 2019-01-09:
yes, i think this is from my xlite (open tx) or from betaflight...
I just bought a brand new mobula7 with crazybee board (frsky SPI), my connection to the micro quad no reliable, but I use my X9D plus, is just normal...
Dont you think it is from opentx?
Or I did miss something?

Thanks

3djc, 2019-01-09:
No. OpenTx does NOT 'do' the radio communication. It only send values to the module, saying, send this value to ch1, this other value to ch2,... How it is done is decided by the module itself. So anythiing abnormal range;, unreliable link, sudden loss, all of that is not OpenTx

slyers77, 2019-01-09:
okay....
after I upgraded to 2.2.3 (I Just did), my connection seems better...
I guess it fix at 2.2.3...

thanks

kilrah, 2019-01-09:
There has been no change with anything communication related in 2.2.3.

slyers77, 2019-01-11:
Yes... you're right.
My problem solved when I switched to D8 protocol.
I've read at the other forum, that this problem occur in f411rx board with d16 protocol.
So their suggestion is to use D8 protocol, instead of D16.

Thanks for your infos