opentx/opentx #6284: Horus timer widget:display only timer name if defined.

Oldgibber, 2019-01-02:
See closed issue #5684 
This bug was corrected for the transmitter software and the Companion simulator but not for the stand-alone simulator. 
[image]

Oldgibber, 2019-01-03:
You are right, as always. I needed to update the sim. However, could you add a **Help** or **About** tab to the stand-alone sim to enable one to check the version number, or perhaps add this to the bottom of the Help menu. The only way I could find out was to download again.

kilrah, 2019-01-03:
Navigate to the version screen in the simulated radio. Standalone simulator is installed together with companion so should always be the same version unelss you do weird stuff with multiple installs...

closed 2019-01-03 by kilrah