opentx/opentx #6287: CRSF Lua script doesn't fit QX7 screen

nmaggioni, 2019-01-06:
Issue #5165 is stale as the script has already been ported to the QX7, but its UI is terrible: the different display size from the X9D wasn't accounted for and most values go off screen to the right.

The device script should be revisited to give each value the most screen space available next to the key name - values such as the MAC address are really tough to fit on the X7's screen, just like longer Pilot or Craft names. In the meantime #6286 fixes the initial loading screen.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the `crossfire.lua` script.
2. Connect a Crossfire-compatible module.
3. Enter one of the devices.
4. Notice that values overflow the screen.

**Screenshots**

[image]

[image]

**General information**
 - OpenTX v2.2.3 N683
 - FrSky Taranis QX7

qral, 2020-03-25:
Still valid.

[image]
[image]

jforhan, 2020-07-21:
Looking at the scripts real quick, I don't believe it is a function of the script. The text strings are received from the TBS module then displayed. Since I don't think you can scale the font on the Taranis LCD screen there is no easy way to do it.

Even my Horus X10s Express has issues. Some of the text is cut off. That may be a script issue of not getting the entire string from the TBS. (I.e. "Do not interru" vs "Do not interrupt" ). There are also other issues with the X10s Express like random characters when RX is updating and non-consistent display updates.

Jeff

closed 2021-06-04 by stale[bot]