opentx/opentx #6290: OpenTX 2.2.2 issue with Amber Sound Pack 2.2  - SDcard access blocked

EricP17, 2019-01-07:
OpenTX version 2.2.2
Hardware: Taranis X9D+
Sound Pack: Amber 2.2
Language: English
Frequency: 100%

Scenario

You are using a countdown timer to monitor the flight time (Timer 1 in my case).  The countdown timer is set to trigger the countdown on a Throttle function.  In my case Ths.  I also have sound events associated to switches.  In particular I have a throttle enable/disable switch as a safety measure which not only disables the throttle stick but also has an audio announcement.

In normal operation the throttle enable/disable switch announces that the throttle stick is enabled/disabled.  However, after the countdown timer expires the audio message from the throttle enable/disable switch no longer functions.  Furthermore ALL access to the SDcard is blocked.  i.e. You cannot browse the SDcard on the Taranis.  The throttle enable/disable switch still functions to enable/disable the throttle, only the audio message is broken because the system cannot access the SDcard.  The issue does not occur when using a switch to trigger the timer, only a throttle function.  The issue does not occur using the companion simulator.  The issue does not occur when using the standard 2.2V0018 SDcard files. There are no known adverse effects to the flight controls.

Expected behavior.

Sound and SDcard access should continue to work after the timer has expired.

Possible workarounds/solutions

The audio files for the 3 timer expired messages are on the SD card in the following directory.

sounds\en\system\timovr1.wav
sounds\en\system\timovr2.wav
sounds\en\system\timovr3.wav

Workaround

rename the files to

sounds\en\system\remtimovr1.wav
sounds\en\system\remtimovr2.wav
sounds\en\system\remtimovr2.wav

This just prevents the files being played so you do not get a timer expired message.  Not really a problem when it is giving beeps every second for the last 10 seconds.

Fixes

Copy the original Opentx 2.2.2  (2.2V0018) 3 system files to the SDcard overwriting the ones from the Amber 2.2 sound pack. 

[amber22.zip](https://github.com/opentx/opentx/files/2734022/amber22.zip)

EricP17, 2019-01-07:
Attaching config file for easy repro. Rename file extension to otx.  Timer is set to 7 minutes but can be shortened for repro.

[config.zip](https://github.com/opentx/opentx/files/2734096/config.zip)

3djc, 2019-01-07:
Well, That’s a great post, unfortunately we are not either authors or maintainer of the Amber pack !

EricP17, 2019-01-07:
Yes I thought you might say that.  I couldn't find anywhere on the internet to report the bug directly to Amber.  I was hoping you had a path.  I still would not have expected OpenTX to block access to the SDcard no matter what.  Thanks for the prompt response

3djc, 2019-01-07:
I’ll have a look at that aspect of your report, just to double check there isn’t an opentx bug there worth solving

kilrah, 2019-01-08:
Most likely it is an SD card issue, try to format it and copy contents again, or even replace the card.

Too many people using that combo for it to be likely to be an issue in OpenTX...

notlebowski, 2019-01-08:
I agree...
Another user recently had a similar issue and a format/copy solved it.
Best to completely replace the card....
By renaming the files and then copying other wavs on it the files will be placed on another part of the SD-Card.

I still think it's an improvement to use a natural voice wherever possible in this set, so I'm glad I redid those 3 wavs

EricP17, 2019-01-08:
Okay here is what I did.

1. Download the amber code again from  https://hmvc.eu/Amber22.rar.  I noticed that this amber22.rar was different even though it had the same name.  I suspect that this is normal.  It looked like the contents had just been modified today so I decided to change as little as possible for debug and use my original file.  You should not be concerned that the version I uploaded was a zip file instead of a rar because GitHub does not accept rar files so I re-zipped the files.  I believe I got my amber22.rar originally from your link but cannot be sure now.
2. I renamed the 3 working files from the Opentx 2.2.2 (2.2V0018) which were currently on the SDcard and then copied the corresponding amber files from my original amber22.rar.  It started to work correctly.

3. I then renamed those working files on the SDcard to remworktimovr1, remworktimovr2, remworktimovr3 and then re-named the failing files back to timovr1, timovr2, timovr3.  It failed again.

4.  I then ran chkdsk against the SDcard.  No issues found and the sound issue was srill there..

5. I then switched from using THs to trigger the timer to using switch A and still got the issue.  Previously the issue did not occur when using a switch to trigger the timer.

6. I tried comparing the contents of the wav files using FC (File Compare) but this proved futile since the files appear to get modified every time you play them.

7. I then used the browser on the Taranis to play the timovr messages directly.  The results were interesting.  The timovr1 message failed 100% of the time and was also cut short although by varying amounts.  The timovr2 message failed about 1 in 4 and the timovr3 message I could not get to fail.

The SDcard is the one that shipped with the system and is unbranded.  I will switch to a different SDcard since I don't trust this one even if I reformatted it.

I did not bother with the re-format test since simply copying the files onto a different part of the flash was sufficient.

kilrah, 2019-01-08:
> I renamed the 3 working files from the Opentx 2.2.2 (2.2V0018) which were currently on the SDcard and then copied the corresponding amber files from my original amber22.rar. It started to work correctly.
I then renamed those working files on the SDcard to remworktimovr1, remworktimovr2, remworktimovr3 and then re-named the failing files back to timovr1, timovr2, timovr3. It failed again.

OK, so it's definitely a card problem.

closed 2019-01-08 by kilrah

randomstash, 2019-02-17:
So was this an SDCard issue finally? I have the exact same problem with a fresh Amber install from today.

3djc, 2019-02-17:
Yes, SDCard issue

EricP17, 2019-02-17:
The bottom line is after changing the SDcard I have had no further issues.

For the inquisitive I did perform one other test which was to copy the bad sdcard (sector by sector) to a new SDcard.  The clone was done on a PC using Ghost.  The copy worked.  The conclusion that this is some kind of timing problem with that SDcard and the Taranis which occurs when reading certain areas of the card.  Note that I was able to read that card correctly on my PC to make the clone on a different SDcard and the clone on the different SDcard works in the Taranis.  I had to clone to a larger SDcard since I did not have any spare 1GB cards around.

choppergirl, 2021-02-05:
I just released (2/4/2021) a new "Edgy" Sound Pack for Open TX transmitters.

❤️️ Some neat things about my Sound Pack: ❤️️
--Drop in replacement for other English Sound Packs
--750 new audio files, more than any other sound pack
--Unlike other Sound Packs, you can browse in Windows Explorer and see the phrase said in the file, and sort by Album / Category. Yes! Godsend!  See screen shot of directory here: [](http://air-war.org/graphics/edgy_dir_listing.jpg)
--No more 4 different voices on your radio, finally your radio will talk to you in one voice.
--I down sampled everything to 16kbs, with no loss in sound quality, which should work on even the oldest transmitters, but I only have TX-16s so no way to test.
Anyone with older or different radios who could confirm my sound files play on it, and let me know, that would be gang bangers.

Short demo on Youtube: [](https://www.youtube.com/watch?v=wA2ARXWUYgw)

On Youtube: [](http://youtube.com/choppergirl)

Project Home [](Page: http://edgy.air-war.org)

Direct Download Link :  [](http://air-war.org/edgy/Choppergirls%20Edgy%20SoundPack%20English%20for%20OpenTX.zip)

Download Dir: [](http://air-war.org/edgy/)

Read Me File: [](http://air-war.org/edgy/_ReadMe.txt)

Rock a New Girl on your Radio :-)
 

█ ♞█ CHOPPERGIRL @ AIR-WAR.ORG ❤️️
☁ Choppergirl's ✈ Flying ✈ Circus ☁