opentx/opentx #6293: Make it possible to multiply with a constant in the telemetry's calculated field

notlebowski, 2019-01-10:
When measuring the voltage of the battery pack with A2 I'm using a trick to get the average cell voltage.
I copy the A2-line which gives me the full voltage and use that copy with a different ratio.

When giving voltage warnings, I prefer to hear typical LiPo values like 3.7V instead of 11.5V which I have to divide in my head while flying.
That's working.

I can't however use this trick with the VFAS-sensor and the FrESC32_80A
I then thought I could use a calculated field and use a multiply by 0.25 (for a 4S), but I then found out I can only multiply 2 sensor values. No gvars or constants.

It would be nice if it could do that as well so I can get average cell voltage with those sensors as well.

closed 2019-02-25 by notlebowski

ahmad-kassem, 2021-04-07:
notlebowski Did you find a solution for this?

Woita2095, 2021-04-23:
I would like to re-open it, such feature would be very useful.
Thanks!