opentx/opentx #6294: if (type(gpsLatLon) == "table") always returns TRUE with OTX 2.2.2

BalsaBasher, 2019-01-10:
On OTX 2.1.9 the following LUA code (taken from OpenTx LUA reference guide) returns FALSE until the GPS sensor is ready, but on OTX 2.2.2 it returns TRUE with zero values in the table until the GPS sensor is ready. Is this intentional or a bug ?

local function background()
  gpsLatLon = getValue(gpsId)
  if (type(gpsLatLon) == "table") then
    gpsValue = rnd(gpsLatLon["lat"],4) .. ", " .. rnd(gpsLatLon["lon"],4)
  else
    gpsValue = "not currently available"
  end
end

The build is opentx-x9e-shutdownconfirm-eu-noheli-lua-luac-sqt5font-en

3djc, 2019-01-11:
As per documentation on getValue ;

value current source value (number). Zero is returned for:
non-existing sources
for all telemetry source when the telemetry stream is not received

closed 2019-01-11 by 3djc

BalsaBasher, 2019-01-14:
If that is the way it is going to be in 2.2.2 then so be it, but it is a change form the way it worked in 2.1.9 and the new behaviour does not match the example LUA code in the 2.2.2 LUA guide.

From: 3djc 
Sent: Friday, January 11, 2019 6:03 AM
To: opentx/opentx 
Cc: BalsaBasher ; Author 
Subject: Re: [opentx/opentx] if (type(gpsLatLon) == "table") always returns TRUE with OTX 2.2.2 (#6294)

As per documentation on getValue ;

value current source value (number). Zero is returned for:
non-existing sources
for all telemetry source when the telemetry stream is not received

3djc, 2019-01-14:
Help is more than welcome, feel free to PR better/more up to date exemples to the doc !