opentx/opentx #6304: Fail to boot 2.2.3 on x-Lite

meoso, 2019-01-19:
**Describe the bug**
Power-On (Boot) fail after using companion 2.2.3 to install latest firmware (2.2.3) to FrSky X-Lite.  It would just lcik some speaker sounds, but would not show the 4 boot squares.

**To Reproduce**
Use companion 2.2.3 to update x-Lite

**Expected behavior**
Normal operation

**Screenshots**
n/a

**General information**
 - OpenTX version: 2.2.3
 - Hardware: FrSky x-Lite
 - OpenTX Companion: 2.2.3 64 bit .deb on Ubuntu

Repaired by downloading and applying the FrSky version from https://www.frsky-rc.com/taranis-x-lite/

kilrah, 2019-01-20:
What version were you on before? Are you holding power long enough? What method used for upgrading? Updating while in bootloader in mass storage mode can fail on linux.

SchonRichtig, 2019-01-22:
I`ve used Windows 10 to update to 2.2.3 without any problems on my X-Lite

meoso, 2019-01-24:
@3djc resultant file:`opentx-xlite-noheli-lua-luac-en.bin`, 2.2.3 (ae5ac05b) variant 218 2019-01-07 10:27:23 OFF with hardware compat check

@kilrah prior was `opentx-xlite-noheli-luac-en.bin` 2.2.2 (a5de713b) variant 218 2018-07-31 13:45:40 OFF -- i see it did not have basic `lua`.  i will try again. during the weekend.

3djc, 2019-01-24:
Somebody reported an issue on mint (an Ubuntu variant) when they activated check hardware compatibility. Could you try unchecking it ?

meoso, 2019-02-03:
i tried without "check hardware compatibility" and it **_worked_**.  did not do extensive testing.

3djc, 2019-02-03:
Ok, so we have an issue with recent Ubuntu. Will try to have a look

3djc, 2019-02-04:
@mpaperno if you are arround, could you get in touch, this one could use your skills :)

bsongis, 2019-08-17:
Please reopen an issue if the problem is still there in 2.3

closed 2019-08-17 by bsongis