opentx/opentx #6305: No Flex implementation in 9XR Pro 2.2.3

L4ky, 2019-01-20:
Hi, 
i just flashed my 9XR Pro with 2.2.3.
I selected the option "Flex" then i downloaded firmware.
Firmware name contains "r9mflex".
I flash successfully the firmware.

When i edit my model to apply Flex options i can't see flex option, instead i see old official values.
Options are "FCC" and "EU" instead of "900 MHz" or "868 MHz".
For example, EU 200mW has only "TELEM OFF".

I tried to create a new model, no success.

kilrah, 2019-01-20:
Try to redownload firmware now?

L4ky, 2019-01-20:
I tried multiple times applying "battgraph" too.
No success.

kilrah, 2019-01-20:
If I tell you to do so it's because I changed something...

L4ky, 2019-01-20:
Sorry, i missed "now".

I tried now, redownloading firmware and still no success.
Options enable: battgraph, flexr9m

L4ky, 2019-01-20:
Great, it works! SHA-1 changed, flashed and i see 868 and 915 MHz with their binding options.
Thank you so much!

closed 2019-01-20 by kilrah

PhlJns99, 2019-02-03:
Is this in the nightly builds for 2.2.3? If I select "flex" as an option before downloading, the latest version I get is 2.2.2RC5?

L4ky, 2019-02-03:
No, this issue was in official release 2.2.3.
Try that instead of Release Candidates

3djc, 2019-02-03:
For 9xr pro (and only this one) you need 2.2.4 nightly

kilrah, 2019-02-03:
No, no need, 2.2.3 release is OK. 

You just need to set companion to use releases and not RCs.

commit 00e28d3c (2019-01-20): flexr9m was missing from compile scripts for sky9x (fix #6305)
commit 3b71bdec (2019-01-20): flexr9m was missing from compile scripts for sky9x (fix #6305)