opentx/opentx #6306: Add gps radar on X12S like as frOs

CcLaFleche, 2019-01-20:
Very usefull when the quad is very far and plugged.

3djc, 2019-01-20:
I believe all the pieces required to make one in lua have been there for some time, no sure if someone did make one or not

CcLaFleche, 2019-03-26:
please....

kilrah, 2019-03-27:
Yes, someone did make one already.

CcLaFleche, 2019-03-27:
Where? i Don't find...
Or it's just for display GPS coordinates. I have already this with telemetry.
But more like this, a map  :
[image]

Luznatural, 2020-05-31:
I continue interested on this feature

lshems, 2020-05-31:
PM me on www.justfly.solutions or rcgroups.

Busy with a waypoint flying app, radar should not be difficult to add.

see also https://www.youtube.com/watch?v=ldkvXOGcMv4

Luznatural, 2021-02-12:
I continue interested in this

kilrah, 2021-02-12:
https://github.com/yaapu/HorusMappingWidget

closed 2021-02-12 by kilrah