opentx/opentx #6310: Q X7 Firmware update problem when "check compatibility hardware" is on

mipel85, 2019-01-23:
Hello,
I have done several times the update of my X7 in version 2.2.3 to test the options of compilation. I'm using a X7 profile and everything happens normally unless I check the check box "Check Hardware Compatibility" before starting the transfer. In this case, the screen displays random pixels or the radio turns on and off continuously. This looks like issue # 4821 which describes the same anomaly. The only solution is to disconnect the battery and restart the flashing without checking the box. Why keep this option that seems to generate a major malfunction?
Michel
[image]

kilrah, 2019-01-23:
Could be something causing issues when 2 operations are done through dfu-util in a row? Might want to try older dfu-util versions.

3djc, 2019-01-23:
Clearly not happening here under debian, I'll install a Mint workstation to see if I can reproduce

closed 2019-11-15 by stale[bot]