opentx/opentx #6312: Companion 2.2.3 / 9xr Pro not loading models and settings

sgmservice, 2019-01-25:
I have a recently purchased (used) Turnigy 9XR Pro that when I load 2.2.3 on it I can't get Companion 2.2.3 to open settings an models. 

The radio had an old version of ersky9x on it. I first updated to ersky9x R221 using eepskye Revision 422 and everything seemed to work fine. I then loaded Opentx 2.2.3 (opentx-9xrpro-gvars-autosource-autoswitch-battgraph-sqt5font-multimodule-en.bin) via Companion 2.2.3. The Firmware loaded fine and the radio booted correctly. When I tried to read models and settings from the radio with Companion 2.2.3 I get the progress bar going from 0 to 100% and the Close button becomes active but nothing loads and the settings window never opens.

I went back to version 2.1.9 and it all works, Companion loads from the radio and writes the modified settings back to the radio. I then reloaded 2.2.3 opentx-9xrpro-gvars-autosource-autoswitch-battgraph-sqt5font-multimodule-en.bin, opentx-9xrpro-multimodule-en.bin, and opentx-9xrpro-en.bin all with the same results; Companion will not load the settings and models from the radio. See attached "Setting Screen1.jpg" and "Read Screen1.jpg".

Next I loaded nightly build companion-windows-2.2.4N689.exe. When I hit Read Models and Settings from Radio, it loaded SAM-BA but then it did read the data from the radio. Now a new problem: it won't write the data back to the radio. I get an Error "can't write model Base Plane-Open tx on this board doesn't accept this function" followed by "Can't write temporary file". see attached "Write Screen1.jpg" and "Write Screen2.jpg".

Any help would be greatly appreciated. I really need the Multimodule support and don't want to have to do all of the updating thru the radio. and I really prefer Opentx to Ersky9x.

Thanks for your help,
Gene Maurice
[image]
[image]
[image]
[image]

kilrah, 2019-01-26:
With your radio connected to the PC in bootloader mode (with 2.2.3) go to the radio's drive in the file explorer and provide us with the EEPROM file.

sgmservice, 2019-01-26:
Andre,

                                                                  

Here’s the files. Note that everything is still named ERSKY_9X. Should OpenTX have renamed and changed the date modified?

 

 

I really appreciate the response, thanks for your help.

 

Gene Maurice

kilrah, 2019-01-26:
Github mail replies don't support attachments, you need to upload somewhere and give the link.

OpenTX never made a custom bootloader for the 9XR-Pro platform, we just include the one for ersky9x without modification.

sgmservice, 2019-01-26:
Try this:

http://sgmservice.com/Public/

 

Gene

lucajon, 2019-02-04:
got the same problem on my 9xr pro and companion 223, let me know if you solve it.

sgmservice, 2019-02-04:
Some further information. 
When I request Companion to "Read Models and Settings from Radio" and it ends without opening the Models and Setting panel there is a .bin and a .tcl file in c:/users/user/appdata/local/temp folder. This file can be opened by companion and it is an up to date Models and Settings data file. The data can be edited without issue. When I go "Write Models and Settings to Radio" I get an error panel that says "Cannot write model Base Plane - Source??? cannot be exported on this board". Click OK and I get a panel that askes if I want to Write to Transmitter or cancel, click on "Write to TX" and get a progress bar that goes to 100%. When I reboot the transmitter nothing has been updated.

lucajon, 2019-02-05:
hello,
news, i have reinstalled the companion 2.2.3 yesterday, and now its working... at the end of the install the driver from atmel was installed(sam-ba 2.12) automaticaly (this thing don't append in the previous install).
i can now upload the setting, i have to test the other things...
note: i have the two version of companion installed 2.1.9 and 2.2.3 don't know if has somthing to do with this

sgmservice, 2019-02-08:
I too can now read Models and Settings from the radio into Companion but still cannot write the data back to the radion.

sgmservice, 2019-03-27:
Here's the results I'm getting now. 
I can read from the radio with no problem. Once downloaded I can edit, add, delete, fold, spindle and mutilate anything and everything. The problem now is when I go to Simulate or Write to Radio I get the attached errors:

Berto105s, 2019-08-24:
I have similar issues with companion 2.2.4 working with my Turnigy 9XR Pro. In addition, when I boot the radio I get "SD CARD ERROR, Expected version: 2.2v0021". I loaded the latest SD card contents before loading the latest opentx firmware 2.2.4. I did get an error about SAM-BA missing when trying to write the models I created but saw on youtube that selecting any exe, I picked simulator.exe, would clear the message. Was that wrong, do I need to download? Okay, with all that said, the main issue I'm having is the companion 2.2.4 changes my selected channel order from TAER to RETA. If I try and correct that on the INPUT page I get the same error message listed in the pervious comments, can't write to my radio. Even if I create the model in the 9xr pro and upload into Companion it changes the channel order mention above. I need a little help!

kilrah, 2019-08-24:
The order in inputs doesn't matter, that's not what defines what's on what. It's the mixer that does.

Berto105s, 2019-08-24:
Andre,

When companion 2.2.4 (using the wizzard) creates a model and then saves the
models to my 9xr pro, the aileron and rudder channels are swapped. They're
swapped before they're even written. If I create a model in my radio and
set the inputs and mixer for TAER everything works fine. Once I read them
into companion the inputs get changed to RETA in the INPUTS page.

I know I can specify these changes in the mixer to correct that but why
should I have to? I have a new Jumper T16 and when I create the same model
- the same wizzard - same companion version - same opentx version, the
inputs stay as TAER and everthing work like it should when written to my
radio and in companion.

I appreciate your reply but maybe my explanation will help you better
understand the issues.

Thanks,

Kevin

kilrah, 2019-08-24:
OK, can reproduce. Was OK on saving and the things get mixed on reading.

[image]

It was not clear becasue as mentioned the order of inputs doesn't matter, but you didn't say the source didn't match the input name anymore.

Berto105s, 2019-08-24:
That's it! That's exactly what I'm seeing. Sorry for the confusion.

Berto105s, 2019-09-08:
Andre,

Is this problem being worked on for the next version of Companion?

Thanks,
Kevin

kilrah, 2019-09-08:
Nobody has taken on working on this at this point, but it's been added to the long list of things to do.

Berto105s, 2019-09-08:
Okay, I'm new to the forum so I wasn't sure how these problems get solved.

Thanks Andre!

gweep, 2019-09-14:
> OK, can reproduce. Was OK on saving and the things get mixed on reading.
> 
> [image]
> 
> It was not clear becasue as mentioned the order of inputs doesn't matter, but you didn't say the source didn't match the input name anymore.
> 
> Also, closing companion with a 9xr-pro document open seems to make it crash.

This is exactly what I am seeing also...  using 2.2.4 on 9XR Pro.   

[image]

gweep, 2019-09-14:
A little more info on this issue:  [OpenTX 2.2.4/9RX-Pro]   Apparently this version of OpenTX has a fixed Input/Control Surface assignment for the Input Tab.

My in-elegant solution:     
* On the Input Tab - Don't fight OpenTX; go ahead and Switch Rudder and Elevator.  
* On the Mixes Tab - correct the input selections while maintaining the correct channel order.   
* Don't forget to swap your rudder and aileron switches.   
* In order to stop P1 from being swapped with P2 with P1, I had to move my I6:Flap up to the I5:Flap position.  

No sense in fighting mother nature.  

[image]

[image]

Berto105s, 2019-09-14:
So you're suggesting that I should go back through all of the models I have
created and do the aileron/rudder swap on the mixer tab? That's a lot of
work but doable. Isn't ease of managing opentx the main purpose for using
Companion?

Thanks for the suggestion. I do appreciate your looking into this problem.

Kevin

gweep, 2019-09-14:
You will need to review and fix both the Input and Mixes tabs.

Berto105s, 2019-09-14:
I kinda understand now what you suggested after re-reading it. Truth be
told, it would be easier for me to simply create any new models in my radio
and leave my other models as they are. I've had to correct them several
times already before I understood what Companion was doing. Guess I'll save
Companion for my Jumper T16 radio. Everything works correctly there.

Thanks,

Kevin