opentx/opentx #6315: Cannot change Language in openTX companion

exocode, 2019-01-29:
**Describe the bug**
When I open a fresh installation of opentx companion (also deleted preferences), it starts in my system language (german).
**To Reproduce**
Steps to reproduce the behavior:
Download OpenTX companion, install it, open it.
Go to settings change language to anything except german (my default one).
restart Opentx companion

**Expected behavior**
Should change to english, czech or something else than german.

**Screenshots**
[image]

**General information**
 - OpenTX version (e.g. 2.2.3)
 - Hardware (e.g. Frsky X7)
 - OpenTX Companion (e.g. 2.2.3 Apple Mac High Sierra)

kilrah, 2019-01-30:
You're changing the language of the radio firmware you're going to download, not the language of companion itself. That's in the menus.

closed 2019-01-30 by kilrah

etingof, 2019-10-01:
Pardon my possibly silly report... I can't figure out where do I change OpenTX Companion interface language. Somehow it switched to Czech right upon installation, then I can't find where to switch it to English.

I have done the RTFM thing - no luck!

kilrah, 2019-10-02:
It has been fixed on 2.3.
If you want to stay on 2.2 you can follow the steps to delete the preference file in the middle of this: https://github.com/opentx/opentx/issues/6723

etingof, 2019-10-02:
> If you want to stay on 2.2 you can follow the steps to delete the preference file in the middle of this: #6723

Removing `~/Library/Preferences/com.opentx.Companion\ 2.2.plist` does not help! The language is still Czech and the file gets recreated on app restart.

I am able to edit that file though. Is there something I could add/change there to fix the language?

kilrah, 2019-10-02:
See the link that was posted in that thread, Mac OS caches preferences so the file is not the only thing there is. You may need to execute a terminal command to flush the preferences cache, reboot, remove the file, flush the preferences cache again, reboot again.

etingof, 2019-10-02:
Yeah, thank you for following this up!

So far killing `cfprefsd` and removing the plist file does not help either. I am yet to logout/reboot etc.

Though  do not understand what's the ultimate desired state of the Companion app... Do we need it to have something specific in the config? Every time I launch it after the above mentioned clean up, it starts anew i.e. offering updates, radio configuration etc.

In other words, why does it use Czech? If it's the default, should I change that to something else in the config?

kilrah, 2019-10-02:
OK then if it starts fresh removing preferences worked.
There is a bug due to a duplicate translation that causes QT to break, if you select CZ the language setting menu disappears and you can't get out of it.
There's no reason it should choose it on its own though, so removing preferences has always worked to get back to the "system language" default setting.

The preferences file is binary but should have "cz" somewhere, if yes then changing that to "en" might work.

etingof, 2019-10-02:
> OK then if it starts fresh removing preferences worked.

I am out of luck!

> The preferences file is binary but should have "cz" somewhere, if yes then changing that to "en" might work.

I use 'TextWrangler' tool from AppStore to read/edit this `.plist` file. 

Here's the contents of that file in XML representation. Just in case you happen to know the right option there. ;-)

Thanks!

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>OpenTxBranch</key>
	<integer>0</integer>
	<key>Profiles.profile0.Beeper</key>
	<string></string>
	<key>Profiles.profile0.ControlNames</key>
	<string></string>
	<key>Profiles.profile0.ControlTypes</key>
	<string></string>
	<key>Profiles.profile0.Display</key>
	<string></string>
	<key>Profiles.profile0.GSStickMode</key>
	<integer>0</integer>
	<key>Profiles.profile0.Haptic</key>
	<string></string>
	<key>Profiles.profile0.Name</key>
	<string></string>
	<key>Profiles.profile0.PPM_Multiplier</key>
	<integer>0</integer>
	<key>Profiles.profile0.Speaker</key>
	<string></string>
	<key>Profiles.profile0.SplashFileName</key>
	<string></string>
	<key>Profiles.profile0.StickPotCalib</key>
	<string></string>
	<key>Profiles.profile0.TimeStamp</key>
	<string></string>
	<key>Profiles.profile0.TrainerCalib</key>
	<string></string>
	<key>Profiles.profile0.VbatCalib</key>
	<integer>0</integer>
	<key>Profiles.profile0.VbatMax</key>
	<integer>0</integer>
	<key>Profiles.profile0.VbatMin</key>
	<integer>0</integer>
	<key>Profiles.profile0.burnFirmware</key>
	<false/>
	<key>Profiles.profile0.countryCode</key>
	<string></string>
	<key>Profiles.profile0.currentCalib</key>
	<integer>0</integer>
	<key>Profiles.profile0.default_channel_order</key>
	<integer>0</integer>
	<key>Profiles.profile0.default_mode</key>
	<integer>1</integer>
	<key>Profiles.profile0.fwName</key>
	<string></string>
	<key>Profiles.profile0.fwType</key>
	<string>opentx-x9d+</string>
	<key>Profiles.profile0.pBackupDir</key>
	<string></string>
	<key>Profiles.profile0.penableBackup</key>
	<false/>
	<key>Profiles.profile0.rename_firmware_files</key>
	<false/>
	<key>Profiles.profile0.sdPath</key>
	<string></string>
	<key>Profiles.profile0.simulatorOptions</key>
	<data>
	QFZhcmlhbnQoAAAAfwAAABFTaW11bGF0b3JPcHRpb25zAAACAMO/w7/Dv8O/w7/Dv8O/
	w7/Dv8O/w7/Dv8O/w7/Dv8O/w7/Dv8O/w78AAAAAAMO/w78AAAAAAAAAAMO/w7/Dv8O/
	w7/Dv8O/w7/Dv8O/w7/Dvyk=
	</data>
	<key>Profiles.profile0.vBatWarn</key>
	<integer>0</integer>
	<key>Profiles.profile0.volumeGain</key>
	<integer>10</integer>
	<key>appDebugLog</key>
	<false/>
	<key>appLogsDir</key>
	<string>/Users/user/Documents/OpenTX/DebugLogs</string>
	<key>arm_mcu</key>
	<string>at91sam3s4-9x</string>
	<key>avr_arguments</key>
	<string></string>
	<key>avr_port</key>
	<string></string>
	<key>avrdudeLocation</key>
	<string></string>
	<key>backLight</key>
	<integer>0</integer>
	<key>backupOnFlash</key>
	<true/>
	<key>backupPath</key>
	<string></string>
	<key>checkHardwareCompatibility</key>
	<true/>
	<key>compareWinGeo</key>
	<data>
	</data>
	<key>dfu_arguments</key>
	<string>-a 0</string>
	<key>dfu_location</key>
	<string></string>
	<key>embedded_splashes</key>
	<integer>0</integer>
	<key>enableBackup</key>
	<false/>
	<key>fwTraceLog</key>
	<false/>
	<key>fwserver</key>
	<integer>0</integer>
	<key>gePath</key>
	<string></string>
	<key>generalEditTab</key>
	<integer>0</integer>
	<key>history_size</key>
	<integer>10</integer>
	<key>icon_size</key>
	<integer>2</integer>
	<key>js_ctrl</key>
	<integer>0</integer>
	<key>js_support</key>
	<false/>
	<key>lastDir</key>
	<string></string>
	<key>lastFlashDir</key>
	<string></string>
	<key>lastImagesDir</key>
	<string></string>
	<key>lastLogDir</key>
	<string></string>
	<key>lastUpdatesDir</key>
	<string></string>
	<key>libraryPath</key>
	<string></string>
	<key>locale</key>
	<string></string>
	<key>mainWindowGeometry</key>
	<data>
	AdnQywACAAAAAANWAAABygAACf8AAAWfAAADVgAAAeAAAAn/AAAFnwAAAAAAAAAACgA=
	</data>
	<key>mainWindowState</key>
	<data>
	AAAA/wAAAAD9AAAAAAAABn4AAAOCAAAABAAAAAQAAAAIAAAACPwAAAACAAAAAAAAAAEA
	AAAKAFcAcgBpAHQAZQMAAAAA/////wAAAAAAAAAAAAAAAgAAAAMAAAAIAEYAaQBsAGUB
	AAAAAP////8AAAAAAAAAAAAAAAgARQBkAGkAdAEAAAJP/////wAAAAAAAAAAAAAACABI
	AGUAbABwAQAAAlz/////AAAAAAAAAAA=
	</data>
	<key>maximized</key>
	<false/>
	<key>mcu</key>
	<string>m64</string>
	<key>mdiWinGeo</key>
	<data>
	</data>
	<key>mdiWinState</key>
	<data>
	</data>
	<key>modelEditGeometry</key>
	<data>
	</data>
	<key>newModelAction</key>
	<integer>1</integer>
	<key>outputDisplayDetails</key>
	<false/>
	<key>profileId</key>
	<integer>0</integer>
	<key>programmer</key>
	<string>usbasp</string>
	<key>recentFileList</key>
	<array>
		<string></string>
	</array>
	<key>removeModelSlots</key>
	<true/>
	<key>samba_location</key>
	<string></string>
	<key>samba_port</key>
	<string>\USBserial\COM23</string>
	<key>settings_version</key>
	<string>220</string>
	<key>show_splash</key>
	<true/>
	<key>simuDbgFilters</key>
	<array>
		<string></string>
	</array>
	<key>simuLastProfId</key>
	<integer>-1</integer>
	<key>simuSW</key>
	<false/>
	<key>snapshot_to_clipboard</key>
	<false/>
	<key>snapshotpath</key>
	<string></string>
	<key>startup_check_companion</key>
	<true/>
	<key>startup_check_fw</key>
	<true/>
	<key>tabbedMdi</key>
	<false/>
	<key>theme</key>
	<integer>1</integer>
	<key>warningId</key>
	<integer>0</integer>
</dict>
</plist>
```

kilrah, 2019-10-02:
<key>locale</key>
<string>**en**</string>

etingof, 2019-10-02:
Hey, changing `locale` to `en` followed by `killall cfprefsd` solved this! Už žádné české!

@kilrah rocks! Thank you!

BlackBird11, 2024-03-06:
i am also trying to change the language of the program on macos. unfortunately there is no com.opentx.Companion folder in preferences
any help is appreciated!