opentx/opentx #6316: Firmware upgrade Fails

cbeith, 2019-01-30:
Hi there am trying to upgrade to the latest firmware have not done this for about 18 Months. On start of Open TX  asks me if I want to download firmware completes successfully but on trying to install get the following message 
[image]

Not sure where I am going wrong any help would be appreciated.

Regards
Chris

bschatz91307, 2019-02-04:
Sorry to bother you, but since I have experiencing the same issue trying to upgrade my X9D (not plus) from 2.2.2 (a5de713b) to 2.2.3, what forum would you recommend?

cbeith, 2019-02-04:
Hi there , 

 

Solution to my issue was to setup the Radio profile in companion. The OpenTX forums have always been helpful.

 

Regards

Chris