opentx/opentx #6318: increase maximum number of mixers

Tiefflieger68, 2019-02-01:
Please increase maximum number of mixers. At the moment it is possible to create max 64 mixers.
In some configurations at multi function models I ran against the limit of maximum mixers.
The Horus Radios should have the ressources to handle more mixers. Hopefully the it will work with the Taranis Radios too.

Tiefflieger68, 2019-02-01:
of cause in some cases one input could replace one mixer. But the inputs are limited too. At the moment I use this option as workaround. My idea is that the new radios should have the ressouces to handle more mixes.

Tiefflieger68, 2019-02-01:
I own DIY switch modules. Each module has 8 switched outputs, by using only one channel, connected via SBUS. But I need one mixer for each output. Thats the reason why I ran to the limits

Tiefflieger68, 2019-02-09:
one more reason why more mixers is a MUST:
there is an announcement of AndreaHuang in RC-Groups:
https://www.rcgroups.com/forums/showpost.php?p=41065931&postcount=9

> _Together with the new functions that I mentioned above,....., it will support more channels,....._

and more channels cause a need for more mixers

kilrah, 2019-02-09:
Given that both we and the radios we support already allow for 32 channels and almost nobody uses that many it is not really an argument.

closed 2019-08-17 by bsongis