opentx/opentx #6320: (minus)GV's with x.x precision returns 10 times bigger value in mixer

prikrylm, 2019-02-04:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

prikrylm, 2019-02-04:
Example:
add mixer to CH1 as MAX and weight of **-GV1**, where GV1 has set **x.x** precision and value 10 and (for example) conditioned by SA^. 
By swiching SA to ^ channel value becomes -100% (not -10%)
(The same example with +GV1 works fine.)

Bug can be found in Companion simulator X9D+, Horus, and bug is on real X9D+ too.

prikrylm, 2019-02-04:
Oh, sorry for first comment. I left orig text there...

raphaelcoeffic, 2019-07-30:
@prikrylm @elecpower since this issue seems to have been fixed #6321, I guess this issue should be closed, right?

prikrylm, 2019-07-30:
I think yes..

PR #6321 (2019-06-08): Fix negative gvars precision
Fixes #6320
closed 2019-07-30 by prikrylm