opentx/opentx #6327: Curves - 3 char limit

jj-uk, 2019-02-11:
Can we please have more characters for curve names? 3 is just not enough in a helicopter setup to remember what the curves are used for.

5 was ok, really wish this wasn't changed to 3.

urkadurka, 2019-03-10:
Hello, I agree with the request, names like Thr Norm, Pit Norm, Pit Hold are clear, today on helicopter I need to use TNr PNr PHl that are less readable.

jj-uk, 2019-09-06:
stale-bot bump.

Eldenroot, 2019-10-12:
Why it was changed to 3?

notlebowski, 2019-10-12:
Memory..
Other features like the 64 logical switches take extra memory that has to carefully sourced from other places. I'm not a dev, but I appreciate how they are able to extend it each time, but they are at the limit.

closed 2019-10-12 by kilrah

jj-uk, 2019-10-12:
Can you please not close my issue. The limit of 3 chars is a problem for heli pilots.

Who really needs 64 logical switches?

kilrah, 2019-10-12:
>Who really needs 64 logical switches?

Well, who needs more than 3 chars? ;)

Makes no sense to keep it open on its own, nothing will change until the whole storage is overhauled and everything is reconsidered.