opentx/opentx #6328: build opentx under ubuntu 18.4

blairfancy, 2019-02-12:
Hi,
I understand that bug tracker is to report issue but as it wiki is involved in my request it more appropriated to post here.
Can anyone could update wiki to all build opentx under ubuntu 18.04. I can compile without error but when i flash firmware on my radio nothing appear.
cmake command:
`cmake -DPCB=9XRPRO -DTRANSLATIONS=FR -DCMAKE_BUILD_TYPE=Release ../`
build command:
`make firmware`

[CMakeCache.txt](https://github.com/opentx/opentx/files/2857037/CMakeCache.txt)

blairfancy, 2019-02-12:
ok probably the issue is that ubuntu 18.04 come with new gcc-arm-none-eabi package 6.3.1
opentx seems to be not compatible with this version of gcc.

Minerest, 2019-02-21:
Same thing with Betaflight. However the makefile in Betaflight has a command ‘make arm_sdk_install’ that downloaded the correct version. I have no idea where it’s pulling the updated version from tho. If you find out, let us know!

bsongis, 2019-08-20:
I want this one kept open as we will update the compiler soon. Also I would like we have a warning for non supported gcc versions (using a white list)

elecpower, 2019-08-20:
Would also be great if the supported version of QT for Companion is also bumped to latest as there are a number of TODO placeholders in the code waiting for more recent versions. May also remove the need for some code workarounds for QT bugs/deficiencies.

kilrah, 2019-08-20:
The issue with that is that many linux distros provide pretty old versions, even moving to 5.7 got us quite a lot of "I can't run companion anymore"...

elecpower, 2019-08-21:
Though there should be a balance between keeping people happy and what is best for the product.
I agree that the baseline should not be bumped every version upgrade of a dependency but there should be a point/frequency/fundamental benefit where it is for the greater good to bite the bullet.
Topic for lots of debate so I'll leave it there.

closed 2020-05-17 by stale[bot]