opentx/opentx #6329: OpenTX French website

LapinFou, 2019-02-14:
Hi Folks,

I send an email @bsongis few weeks ago about this suggestion.
Since there is a lack of good information for the French community (specially for people who doesn't speak English), I started to create a Mega Topic on the French forum (link is here → [[Méga Tuto] Liste de tous les liens utiles](http://frskytaranis.forumactif.org/t1839-mega-tuto-liste-de-tous-les-liens-utiles)).
A forum is definitely not the best place to write such information and to maintain them.
A French guy (alias **Mipel**) is part of the of the [PHPBoost CMS](https://www.phpboost.com/) team. So we decided to create a "test" website.
The first version is available here (still few graphical things to tweak) → [French OpenTX wiki](http://opentx.ademic85.fr)
We included some articles & news just to show you how it looks.

Now, here is the question:
- In order to guarantee the durability of such website, it would be fantastic if it could be host on the official OpenTX's server. **Mipel** can give you more information about the requirements.
- The main OpenTX members, **Mipel** and myself should have an admin (and FTP) access to this website to do the setup.
- Then, few members would have a restricted access in order to only create, edit articles. This management access is done inside the CMS. 

What do you think?
It would be very cool for the French community. 😄

3djc, 2019-02-14:
I think it is a great idea to have such a site, but no so much under our banner and/or server, for responsibility, fairness with all language (if we do it for french, then surely we have to do it for any language)

Looks to me all you are looking for is hosting really, you can get a vritual dedicated server for 1.99€/month (and you could then have a real forum engine, the one hosting the french site is not the most recent and user friendly one !). Happy to advise on this if needed

LapinFou, 2019-02-14:
"Fairness with all language" is a very good point.
On the other hand, my main concern is the durability of the website. We must ensure that all accumulated information will not be lost if something happen to one of the website's maintainers.

3djc, 2019-02-14:
Do we ? If there is no more mainteners, then this all will quickly become obsolete, and therefore not worth keeping, and given our workload, probably not something we want to commit maintaining should there be no more volunter to keep it alive. (don't get me wrong, it is a fantastic work done there, and I reallty hope it gets maintained, but not something we should comit to maintain is case of no one else wanting too)

notlebowski, 2019-02-14:
If it's multilingual (with a French section) I can make hosting available for free on opentx.nl

LapinFou, 2019-02-14:
@3djc Also a good point.
@notlebowski Thanks you very much for your proposal, however it is not intend to be multi-language. My main regard is not about the cost, but about durability. 😄

notlebowski, 2019-02-14:
I can give you the durability and will treat it like a paid site. With multilingual I mean a site that can go immediately to the French section for users that chose to, with other languages not maintained but could be used if someone would want to.
You would be responsible for the entry and the French section...

LapinFou, 2019-02-15:
@notlebowski Again, thank a lot for you offer. 👍 However the team (including myself) responsible of the creation of this French website does want a host in France. Since the OpenTX team (at least @3djc; I hope @bsongis a @kilrah can give their opinion) cannot host this website for very fair & good reasons, we will choose a professional host service in France.
Have a nice day. 😄

LapinFou, 2019-02-18:
I made my decision (thanks to @3djc), I will create an independent website.

closed 2019-02-18 by LapinFou

3djc, 2019-02-18:
Feel free to ping me if you need help

LapinFou, 2019-02-18:
I will!
Thanks. 👍