opentx/opentx #6332: LUA script kicks me out + BF lua does not show telemetry

OxygenGS, 2019-02-15:
Hello,

I got crossifre since yesterday.
I got a micromodule with version 2.88, and the new Nano diversity receiver and it did bind and update. It also flys amazing, rssi 99! However, i see 2 issues here and cannot fix it.
I use the XQ7 so i did apply the mod with the chip that came with it, also easy. The issues i have are:
1: If i go (like you see in the video) to system -> crossife -> Crossfire lua and execute -> then select the nano diversity receiver -> it opens up but if i go down or even stay idle in it, it just kicks me out and i need to re-open and execute the script. I get no change to change stuff in there, did manage to get rx and tx and the rssi setting in channel 12, but with ALLOT of retrys and this is not how it should work right?
2: Normaly with my X4R i did hold page in the main screen to get the PID and stuff via de BF lua. Now however if i open it, it does not show me any valleus, how to fix this pls? this is a big issue out on the field when i wanna tune my quad and chnage settings.

I hope you guys have a fix for me for these 2 issues.d

Greets
[Video.zip](https://github.com/opentx/opentx/files/2870802/Video.zip)

3djc, 2019-02-16:
The sympthoms you describe suggest you have issues getting telemetry (possibly in both direction). Sound like an hardware issue (have you tried lowering the speed in hardware menu ?).

Anyhow, this does not seem to be an OpenTx bug

OxygenGS, 2019-02-16:
What do you mean with lowering the speed in the hardware menu??

3djc, 2019-02-16:
[image]

OxygenGS, 2019-02-16:
it is on 4000000 now, to what should i set it?

OxygenGS, 2019-02-16:
https://ufile.io/0x8pq -  here a video of issue: 2
Did even update the micro to several diffrent version but nothing did help

3djc, 2019-02-16:
You really should contact tbs, to find out what is defective in your setup

kilrah, 2019-02-16:
I've seen a few reports of issues with the diversity Nano RX. Maybe something changed with it and requires other stuff to adapt but we know nothing about it.

OxygenGS, 2019-02-16:
Ow ok, jeah i am already in contact witht hem. I hope it's just a firmware issue with the new diversity receiver that can be easily fixed

3djc, 2019-02-16:
If they give you detailled information about things we need to change to fully support those receivers, we will be happy to do those changes (time allowing, obviously)

OxygenGS, 2019-02-16:
I will keep you guys updated.

OxygenGS, 2019-02-16:
update: a friend of myn had 1 mod chip spare. So i did solder it to it, and still the same issue: no telemetry screen in the BF LUA.

OxygenGS, 2019-02-16:
also ALL steps on this website: https://oscarliang.com/crossfire-betaflight/ i did follow.

Still it's not working in the BF lua and still kicks me out of the crossfire LUA

OxygenGS, 2019-02-17:
Ok update:

Today i did go to a friend and he has the xq7 also modded and also crossfire.

I did connect my taranis to his build, and yes! i can see the BF lua script values and also the lua script does not crash on me! So now i do know: it's not my module, it's also not the taranis and the mod chip is working. It's now either the flight controller or the receiver that does not work as it should. However, i do fly already 2 months with this fc with the X4R previously installed, and it works perfectly. So my guess is: the receiver is not working like it should. What should i do now? I am still waiting on response from TBS

3djc, 2019-02-17:
Well now it is confirmed it is not OpenTx software the issue , so I’m closing this.

closed 2019-02-17 by 3djc