opentx/opentx #6334: Altitude not logged into SD LOGS even though enabled

wx4cb, 2019-02-17:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
Have crossfire, inav sensors all checked for logging, the only option that I can see that is not logged is altitude.

**Expected behavior**
Altitude is logged along with other data

**Screenshots**
These are the values ;logged in the CSV file from SDLOGS being on 
Date | Time | 1RSS(dB) | 2RSS(dB) | RQly(%) | RSNR(dB) | RFMD | TPWR(mW) | TRSS(dB) | TQly(%) | TSNR(dB) | GPS |   | Sats | Ptch(rad) | Roll(rad) | Yaw(rad) | FM | RxBt(V) | Curr(A) | Capa(mAh) | GSpd(mph) | Hdg(@) | Rud | Ele | Thr | Ail | S1 | S2 | S3 | LS | RS | SA | SB | SC | SD | SE | SF | SG | SH | LSW | TxBat(V)

one drive link to CSV file: https://1drv.ms/u/s!AtPOuMhazR4PgWrpllKB1nPthmhD

**General information**
 - OpenTX version 2.2.3
 - Hardware Frsky X9D+ SE
 - INAV 2.1 RC3

3djc, 2019-02-17:
Well we cannot replicate since we do not have crossfire gear, but sensor logging is a very generic function there is no reason for a specific sensor to be omitted, and alt logging works fine with frsky, multi,...

Is the sensor working properly beside logging ? Have you tried logging just this one ?

wx4cb, 2019-02-18:
havent tried with just that one, but yes it works perfectly fine with telemetry screens (num, bars etc) and also the inav telemetry script. and also with "play values" in special functions.

it also isn't logging it when i use the frsky gps sport sensor - tried that today on a different one model

kilrah, 2019-02-18:
There is no limit.

BTW why are you using an old RC version?

3djc, 2019-02-18:
Just tried that with FrSky GPS sensor, seems to work fine here

wx4cb, 2019-02-18:
@kilrah according to companion. there isn't a new version. The first thing i did was to try to upgrade the version but companion says there is no updated version

As you can see from the pictures below, it's definately loggged. and ocmpanion says there's no update avail. i tried updating companion just in case but it's the same version i already have

[image]

[image]
[image]
[image]

teckel12, 2019-02-18:
I run Crossfire and it logs the altitude, here's the header of my log file:

Date,Time,FM,1RSS(dB),2RSS(dB),RQly(%),RSNR(dB),RFMD,TPWR(mW),TRSS(dB),TQly(%),TSNR(dB),RxBt(V),Curr(A),Capa(mAh),GPS,GSpd(mph),**Alt(ft)**,Sats,Ptch(rad),Roll(rad),Yaw(rad),Rud,Ele,Thr,Ail,S1,6P,S2,LS,RS,SA,SB,SC,SD,SE,SF,SG,SH,LSW,TxBat(V)

@wx4cb You may want to download the OpenTX companion 2.2.3.  Version 2.2.3 of OpenTx is available.  It's possible running a release candidate is causing your problem.

wx4cb, 2019-02-18:
ok this is weird...... i completely cleared the logs directory of all logs and now it's logging altitude however on two different aircraft there are parts of the logs where it's just the same data over and over again on the one aircraft it logs the same altitude and speed etc (the speed btw is way off) for quite a considerable amount of time. there is obviously no way that the aircraft (a flying wing) could hover in position.

Could this be a gps issue? i don't think it is, as it's a brand new gps and the OSD was working fine with regards to posit/alt and all the other parameters (from memory)

[image]

wx4cb, 2019-02-18:
@teckel12  i will do that. but normally companion says.. hey there's a new version out but it hasn't done that

notlebowski, 2019-02-19:
There is no new 2.2.2 release
You need to install 2.2.3 and then select nightly builds or stay at 2.2.3 
nightly builds don't go to the release nor to the next nightly build

kilrah, 2019-02-20:
You have at some point set companion to download release candidates, set it back to download stable releases. 

If you enable logs for a new sensor after already having logged some stuff that day you need to delete the existing log for that day or the column header for the newly loged sensor will be missing, so it won't be seen.

closed 2019-02-20 by kilrah