opentx/opentx #6338: X9D+ 3 Position Switch for SF

HoyDAviDJ, 2019-02-20:
I recently changed the switches in my X Lite to all 3 position switches.  In OpenTX Companion you can change the switches to 3 position so that OpenTX recognizes all the positions.  This can also be changed in the menus of he transmitter.

I then wanted to change SF on the X9D+ from a 2 position to a 3 position.  However, when I went to see about changing it the only options are a 2 position, 2 position toggle or none.  All of the other switches can be changed from 3 position to 2 or toggle, yet SF cannot be more than a 2 position switch.

Is there a hardware reason for this that I am unaware of or is this possible to have changed so that SF can be a 3 position switch?

3djc, 2019-02-21:
There is only one pin for SF, therefore you cannot put more than 2 pos

closed 2019-02-21 by 3djc

HoyDAviDJ, 2019-02-21:
Ahh, that explains it.  Thank you