opentx/opentx #6339: Start one-time LUA script on startup/after reset flight for better preflight checklist

prikrylm, 2019-02-22:
Similar functionality as "display checklist", but run one-time LUA script.
User could create own preflight checklist with some interactions, images :)..., there could be some warnings based on persistent timers (i.e. check motor bearings, lube, etc...) and so on.

notlebowski, 2019-02-25:
Wouldn't a run script SF do based on the trigger "one"?

prikrylm, 2019-02-25:
Special functions can start "function scripts" which has not access to screen (non visual). So, trigger "one" is not the solution.

kilrah, 2019-02-25:
No, function scripts do not have LCD access.

lshems, 2019-02-25:
Make an autorun telemetry startup routine.

If there is an Auto.lua defined in the first telemetry screen as Lua script, after exiting the notes, switch to telemetry screen one and execute it.

This solves sope other requests as well.

prikrylm, 2019-02-25:
If I had to switch to telemetry screen, I could run one-time script as well. And switching back to main screen is needed too.

scott-adamson1975, 2019-02-25:
external antenna would be enabled by this point i believe, main requirement of "checklist" for me is to remind me to connect external antenna, would be even better if i could script it to disable for this session if i choose an option.

or the an option for the firmware to always prompt "are you sure you want to  enable external antenna?" if external antenna is enabled for the model (on change and switch on) and based on the response activate or deactivate for this session.

3djc, 2019-02-25:
There is not such thing like disactivate for a session. If your radio where to go emergency mode, you would be back to external which you don't have, here goes a dead craft. It surely sounds like a great idea, much less so when you dig into the details

scott-adamson1975, 2019-02-28:
question regarding emergency mode, and possibly needs explaining in wiki/documentation

when emergency mode activates is there a brief period of time without control while the system shifts to emergency mode?

kilrah, 2019-02-28:
Emergency mode is what the radio gets in after it has crashed/rebooted unexpectedly. So obviously it implies it was non-functional for a whort while.

kilrah, 2019-08-27:
#5156

closed 2019-08-27 by kilrah